freifunk-berlin/firmware #551: Profiles: Remove DNS-Server 213.73.91.35

pmelange, 2018-04-20:
The DNS server 213.73.91.35 is offline.  It is set in many of the profiles and needs to be removed.

https://twitter.com/clubdiscordia/status/983109445784875009?s=11

pmelange, 2018-04-22:
openwrt/luci#1757

pmelange, 2018-04-23:
Pull Request openwrt/luci#1759

pmelange, 2018-04-23:
PR: hnyman merged commit 0c30ee0 into openwrt:master

booo, 2018-04-23:
We should find a replacement for the old one.

The ip is still in the olsrd configuration:

 https://github.com/freifunk-berlin/firmware-packages/blob/master/defaults/freifunk-berlin-olsrd-defaults/uci-defaults/freifunk-berlin-olsrd-defaults#L44

pmelange, 2018-04-23:
Yes, we should find a new one.  A new nameserver should go in its own issue/pr.

The host still responds to pings.  There is no problem with it being in the olsrd ping list

booo, 2018-04-23:
Is there a issue for a new nameserver? Can you create one if there is no issue?

We should not assume that the host stays online and should remove it from the ping list or better replace it.

pmelange, 2018-04-23:
New issue created for a new potential nameserver #552

SvenRoederer, 2018-04-23:
just requested a cherry-pick into lede-17.01 branch (https://github.com/openwrt/luci/issues/1760).

we also need a migratioin for the DNS-server list

booo, 2018-11-03:
This pull request includes a migration script: https://github.com/freifunk-berlin/firmware-packages/pull/152

SvenRoederer, 2018-12-20:
https://github.com/freifunk-berlin/firmware-packages/commit/212db23968edc8bcc5302e9a08a2c51d97490aa0 adds a migration for Hedy-1.0.2

pmelange, 2018-12-20:
Maybe this should be run with every migration. Sort of a standard thing to do, always set the DNS servers. https://github.com/freifunk-berlin/firmware-packages/blob/ea6d78598a3edf4daca5c72208946364b7d1b136/utils/freifunk-berlin-migration/uci-defaults/freifunk-berlin-01-migration.sh#L443-L452

SvenRoederer, 2019-01-13:
master: 
migration: https://github.com/freifunk-berlin/firmware-packages/commit/a2156221f511e4ee4283323dc4b68103b04010bd; profile: https://github.com/openwrt/luci/commit/c2d3fe8a9730a519278def5534c60ee6991185c3#diff-3faaef5bc5179c8442a73059db372a80
V1.0.2:
migration: https://github.com/freifunk-berlin/firmware-packages/commit/212db23968edc8bcc5302e9a08a2c51d97490aa0; profile: https://github.com/openwrt/luci/commit/09ae638e4b4eb6ee11fd718caf209c6c983a6e0e#diff-3faaef5bc5179c8442a73059db372a80

closed 2019-01-13 by SvenRoederer