freifunk-berlin/firmware #552: Profiles: Add a new nameserver

pmelange, 2018-04-23:
A new nameserver is needed for the following profiles in openwrt/luci
profile_berlin
profile_potsdam
profile_cottbus
profile_tulumlibre

The new nameserver shold not log dns requests.

booo, 2018-08-21:
Any suggestions?

For reference the original thicket that introduced the name servers is here https://github.com/freifunk-berlin/firmware/issues/15

Another list of open dns servers: https://wiki.ipfire.org/dns/public-servers

booo, 2018-09-20:
I propose that we add 80.67.169.40 from the French Data Network (FDN) to the list of servers. More information on the website: https://www.fdn.fr/actions/dns/

bobster-galore, 2018-09-20:
Yes, it looks good to me!

pmelange, 2018-09-27:
@booo Did you contact FDN and ask them if it would be ok?  I would start on updating the packages when you get an answer.

pmelange, 2018-10-21:
merged in https://github.com/freifunk-berlin/firmware-packages/pull/145 commit https://github.com/freifunk-berlin/firmware-packages/commit/8d9c50fc411ac3ed51643e5975ad9d5da92ad79b

closed 2018-10-21 by pmelange