freifunk-berlin/firmware #556: Getting runtime errors when trying to access LuCI

mophahr, 2018-05-10:
I run into the following error when trying to open LuCI in the web interface:

```
[msch@fedorabox default]$ curl http://192.168.1.236/cgi-bin/luci

<!DOCTYPE html>
<html lang="en">
	<head>
		<meta charset="utf-8">
		<title>Status: 500 Internal Server Error
/usr/lib/lua/luci/dispatcher.lua:396: Failed to execute function dispatcher target for entry '/'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:396: Failed to execute function dispatcher target for entry '/freifunk'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:396: Failed to execute template dispatcher target for entry '/freifunk/index'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:55: Failed to execute template 'freifunk/index'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'header'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'themes/bootstrap/header'.
A runtime error occured: [string "/usr/lib/lua/luci/view/themes/bootstrap/hea..."]:150: attempt to index local 'boardinfo' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:396: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>
```

According to the ssh login screen, I'm on `Firmware Berlin (%N 2018-alpha rev a0f4533) Generic - ramips/mt76x8` on a gl-mt300n-v2 router.

torte71, 2018-05-10:
Looks like the umask build issue here: https://github.com/freifunk-berlin/firmware/issues/431
Did you build that image on your own? In that case, rebuild with "umask 0022" as stated above.

mophahr, 2018-05-10:
Thanks! I'm currently trying to re-clone the repo with umask set to 0022 and re-building the image. Will report back if that works.

mophahr, 2018-05-11:
I had to switch to the SAm0815_experimental branch due to unrelated issues. There, re-cloning the repo with umask set to 0022 and re-building the image (2018-alpha rev 6070a5d) leaves me with the following error:
```
[msch@fedorabox ramips-mt76x8]$ curl http://192.168.1.236/cgi-bin/luci
/usr/lib/lua/luci/dispatcher.lua:413: Failed to execute function dispatcher target for entry '/'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:413: Failed to execute function dispatcher target for entry '/freifunk'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:413: Failed to execute template dispatcher target for entry '/freifunk/index'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:55: Failed to execute template 'freifunk/index'.
A runtime error occured: /usr/lib/lua/luci/model/uci.lua:157: attempt to call method 'get_all' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:413: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:99: in function </usr/lib/lua/luci/dispatcher.lua:98>
```
I assume this is a different issue than #431, isn't it?

SvenRoederer, 2018-05-11:
just remembering this one ... https://github.com/freifunk-berlin/firmware-packages/pull/129

but had no time to check for yet

SvenRoederer, 2018-05-18:
> I manually applied the change from bd8c09106811ea052b434c68c2bf461c 

I can't identify the commit you mentioned. do you reference to the PR of @nExXxuS ?

mophahr, 2018-05-21:
I was referencing the replacement of `uci.get` with `uci:get` from this PR: https://github.com/freifunk-berlin/firmware-packages/pull/129

SvenRoederer, 2018-05-22:
As PR https://github.com/freifunk-berlin/firmware-packages/pull/129 was merged this should be fixed

closed 2018-05-22 by SvenRoederer