freifunk-berlin/firmware #560: no-tunnel dhcp request for ffuplink veth interface overwrites default gateway for the br-wan bridge

pmelange, 2018-05-11:
When the ffuplink interface goes up (which happens after the br-wan bridge), it requests an IP address from the local net's DHCP server.  Upon receiving this address, the default gateway for br-wan is overwritten.

If the ffuplink address is to be assigned to a the "guest network" (fritz box), then all traffic from the router on behalf of the router, will go out with a src address of the ffuplink device.  This can lead to undesired and additional network traffic as ffuplink.  For example, monitor and DNS traffic.  Here I'm thinking about bandwidth limits imposed on the "guest network".

Another issue is NAT on the telecom router.  If a connection initially has a src from br-wan, then suddenly changes to ffuplink, then the connection breaks since the connection tracking see a different src addr.

As a solution, I suggest that with the "no-tunnel" version, the ffuplink interface sould have a metric != 0.  I've tested with a metric of 1, and it works totally fine.

This may also have a positive impact on the following issues... #547 #532 #542 #545

torte71, 2018-05-13:
I can confirm, that this does fix https://github.com/freifunk-berlin/firmware/issues/545

SvenRoederer, 2018-05-13:
Let's discuss this further in the common issue #561.

closed 2018-05-13 by SvenRoederer