freifunk-berlin/firmware #569: Change berlin.freifunk.net to freifunk.net

bobster-galore, 2018-05-30:
I propose in future we call our network "freifunk.net" which is more common thru freifunk communities and enables users from other communities to use berlin freifunk seamless.
There is no advantage for freifunk users to join a "community-named" ssid.
If someone is interested in the community stuff, it can go to the router starting page where the community (here Berlin) is displayed.

SvenRoederer, 2018-05-31:
Indeed a common SSID used by many communities would be nice for users traveling between different communities.
I heard, that having a short WiFi-scan-list on the mobile-phone will also save some battery-power.

pmelange, 2018-06-24:
On one side, yes a common SSID accress 'schland has it's advantages.  But on the other side, I find it nice that it is also the website for the berlin community.  

Whatever the SSID is, that would be the first place people would go to try to find out more about Freifunk.

From the "freifunk.net" website, it takes 4 clicks (if you know what you are doing) to get to "berlin.freifunk.net".

For this reason, I find the SSID "berlin.freifunk.net" exceptionally good.  Unfortunately the website seems to be unreachable right now.

SvenRoederer, 2018-06-24:
Imho changing the SSID gives no real advantage to the user. But it will require that all clients need to be changed, to join the new SSID.

And esp.for the Berlin-community we will have both SSIDs around for ages .....

booo, 2018-07-05:
I'm in favor of keeping the localized SSID ("berlin.freifunk.net"). I support the arguments from @pmelange and @SvenRoederer.

@bobster-galore Please initiate a discussion on e.g. the mailinglist if you think this is an important topic. Otherwise please close the ticket.

booo, 2018-08-21:
I'm closing this ticket because a discussion was not initiated (as far as I know). Please reopen if this is important to you.

closed 2018-08-21 by booo