freifunk-berlin/firmware #570: Can't download packages list

ikreb7, 2018-05-30:
Hello,

I use the the `TP-LINK Archer C7 v4` I want to up-to-date the packages. So at first i tried to download the latest packages list. However I get this error message:

```
Downloading` http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/targets/ar71xx/generic/packages/Packages.gz
*** Failed to download the package list from http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/targets/ar71xx/generic/packages/Packages.gz

Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/base/Packages.gz
*** Failed to download the package list from http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/base/Packages.gz

Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/gluon/Packages.gz
*** Failed to download the package list from http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/gluon/Packages.gz

Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/luci/Packages.gz
*** Failed to download the package list from http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/luci/Packages.gz

Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/packages/Packages.gz
*** Failed to download the package list from http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/packages/Packages.gz

Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/packages_berlin/Packages.gz
*** Failed to download the package list from http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/packages_berlin/Packages.gz

Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/routing/Packages.gz
*** Failed to download the package list from http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/routing/Packages.gz
```

and

```
Collected errors:
 * opkg_download: Failed to download http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/targets/ar71xx/generic/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/base/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/gluon/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/luci/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/packages_berlin/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/704/packages/packages/mips_24kc/routing/Packages.gz, wget returned 8.
```

If I try to get the files with wget, I get 404. SO I think the paths are wrong, aren't they?

booo, 2018-05-30:
The opkg download url for unstable images/packages is only valid for a certain time. We don't have the resources to keep all unstable builds on the servers.

Take a look at:

http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/

Maybe you can use a newer build as source by changing the build number in the url?

closed 2018-05-30 by booo

ikreb7, 2018-05-31:
I did this. I changed it to [http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/]( http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/) now it said me that the signature is wrong.

```
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/targets/ar71xx/generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/freifunk_berlin_core
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/targets/ar71xx/generic/packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/freifunk_berlin_base
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/packages/mips_24kc/base/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/packages/mips_24kc/gluon/Packages.gz
Updated list of available packages in /var/opkg-lists/freifunk_berlin_gluon
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/packages/mips_24kc/gluon/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/freifunk_berlin_luci
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/packages/mips_24kc/luci/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/freifunk_berlin_packages
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/packages/mips_24kc/packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/packages/mips_24kc/packages_berlin/Packages.gz
Updated list of available packages in /var/opkg-lists/freifunk_berlin_packages_berlin
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/packages/mips_24kc/packages_berlin/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/freifunk_berlin_routing
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/745/packages/packages/mips_24kc/routing/Packages.sig
Signature check failed.
Remove wrong Signature file.
```

Other question. Would it not better, if there would be a button to update the package sources before?

SvenRoederer, 2018-05-31:
You get this as every individual build is using it's own set of sign-keys.
As you are using a unstable development version it's the best to use a more recent build