freifunk-berlin/firmware #571: softbrick of cpe210 v2

SvenRoederer, 2018-06-02:
according to https://lists.berlin.freifunk.net/pipermail/berlin/2018-May/037867.html the current code for support of the TPlink CPR210v2 (https://github.com/freifunk-berlin/firmware/commit/1e1ba9580691fb92178cf6f5fd6beb8869c3192f) leads to soft-bricks.

possibly https://github.com/openwrt/openwrt/pull/937 is a patch with recent working code.

bobster-galore, 2018-06-07:
Do we have to wait for the upstream fix or can we cherry pick?
Has anyone tested if it is working with that pr?

SvenRoederer, 2018-06-10:
It reads as the patch will work.
This patch just needs to replace the current "buggy"

adrianschmutzler, 2018-06-16:
Maybe it's the same problem I had with the initial PR for the CPE210: The config for the CPE210v2 inherits the config from the v1. At some point, the configuration for the v1 was altered to increase the kernel partition size. However, there is a second location where the partition sizes are stored and this one may have the old sizes depending on when you implemented it:

https://github.com/openwrt/openwrt/pull/642#issuecomment-386123966

This should be fixed in the new PR.

However, I'm not sure whether this change will have an impact on the current state of the PR:

https://github.com/openwrt/openwrt/commit/805e59cdcc2b6b2bb3f7583d8ccdb932e6cdeb7b

bobster-galore, 2018-09-13:
Will this be fixed if we rely on the latest openwrt? https://github.com/freifunk-berlin/firmware/issues/583

SvenRoederer, 2018-09-14:
this board should be fixed in OpenWrt-master and OpenWrt-18.06

booo, 2018-11-03:
Please reopen if you encounter problems with this agian.

closed 2018-11-03 by booo