freifunk-berlin/firmware #572: actual failsafe doc link in failsafe mode

bobster-galore, 2018-06-04:
The documentation link in failsafe mode should point to https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset .
http://wiki.openwrt.org/doc/howto/generic.failsafe is depricated.

SvenRoederer, 2018-06-11:
I don't get it ??????

bobster-galore, 2018-06-11:
If u are in failsafe mode on ur router there is a link for more information on failsafe mode, which is deprecated.

SvenRoederer, 2018-06-11:
is it showing the freifunk banner or a openwrt banner?

booo, 2018-06-12:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/base-files/files/etc/banner.failsafe;h=14615e1ef7a08213157f295118fd2a31fb2e502d;hb=HEAD

Could you please file this upstream?

bobster-galore, 2018-06-12:
Done. https://github.com/openwrt/packages/issues/6282

SvenRoederer, 2018-06-25:
fixed upstream for master, 18.06 and 17.01 branches

closed 2018-06-25 by SvenRoederer