freifunk-berlin/firmware #575: What branch should we develop on?

pmelange, 2018-06-24:
@SvenRoederer 
@bobster-galore 
@torte71 
@booo 
@sarumpaet 

It is completely unclear to me where we are developing.  

I would really like to get some things done, but I sit here and look at all the branches and I don't know where to base my work.

The master branch seems to be unreliable. Why?  Shouldn't the master branch work properly?
What is the next_release branch and what is the Heady-1.0.2 branch?  Is there supposed to be a difference?

And the firmware-packages branches are just a confusing.

Is the next release 1.1.0 or 1.0.2?  I see both.  Why?

SvenRoederer, 2018-06-24:
* regarding the planned release  see: https://github.com/freifunk-berlin/firmware/milestones
* master will relate to Hedy-1.1.x
* all other branches are just testing or for features to merge when ready - made by someone

in the packages-repo it's quite the same
* master is the current (supposed to be) working code
* for-Hedy1.0.x can be used for backport from master for the Hedy-1.0.x maintenance releases

pmelange, 2018-06-27:
> * master will relate to Hedy-1.1.x

Shouldn't something that "will be" be a branch which is not master?  But if I understand correctly from your statement, I shouln't base any code off of master if I want to develop for 1.0.2.

> * all other branches are just testing or for features to merge when ready - made by someone

Still, I don't see where to base my code for something to be included in Hedy-1.0.2.  What, when I develop something, should I base my code off of?

SvenRoederer, 2018-07-06:
All goes first on master

closed 2018-07-09 by pmelange