freifunk-berlin/firmware #576: devices with faulty IBSS support

SvenRoederer, 2018-06-25:
There is a growing number of routers around, which do no offer a working adhoc more or even no adhoc anymore.

This ticket is to keep a list of them.

SvenRoederer, 2018-06-25:
* Archer C7 vX
* GlInet mt300n (with OpenWrt master or 18.06)

booo, 2018-08-21:
Can we move this list to the wiki? E.g. to https://wiki.freifunk.net/Berlin:Firmware#WLAN-Router Maybe the openwrt wiki is an even better place to keep this information?

There is a ticket for 802.11s support in the firmware: https://github.com/freifunk-berlin/firmware/issues/536 If we want to address the problem we should keep the discussion in 802.11 ticket.

I'm inclined to close this ticket if there is no objection.

bobster-galore, 2018-09-13:
I think it's not a documentation problem, it's existing in wetworld.
We should switch to mesh(11s) asap and can document that as well and may be issue warnings for new hardware which doesn't support ibss any more.

booo, 2018-09-15:
We have an issue for the switch to 11s: https://github.com/freifunk-berlin/firmware/issues/536

Please document the 11s support of devices in the freifunk or openwrt wiki.

closed 2018-09-15 by booo