freifunk-berlin/firmware #578: [RFC] Integrate "no-tunnel" uplink in the network-defaults

SvenRoederer, 2018-07-08:
I'm planning to integrate the current "uplink-no-tunnel" into the default network-setup, as seen in SAm0815_experimental branch.

* this gives some valid configuration w/o having to include an additional "uplink-package"
* other uplink-types can overwrite the settings they need
* we get rid of an additional package (uplink-no-tunnel) to maintain
* the non-default uplink-types will be prepared to some kind of "I like to change the uplink type"

pmelange, 2018-07-09:
Additional question:
* What would be the impact on the backbone image?

Feedback:
* Keeping the configuration for the notunnel version in the uplink directory seems more intuitive than having to search for it somewhere else (maintenance).
* The ability to have the non-default uplink-types to be interchangable should be possible even if the notunnel version is a package.
* Changing from notunnel to something with openvpn reqires needing to install a lot of dependencies (openvpn, mbedtls, ...) which might be quite difficult to impossible for 4mb devices.  For this reason, I believe that for the time being keeping them as separate images makes sense.   When the tunneldigger uplink type is ready, then changing the uplink types should be realizable for all flash sizes.

SvenRoederer, 2018-07-09:
> What would be the impact on the backbone image?

1. needs to be checked
1. we can disable the autostart of this interface by default on all images. activation we can leave to the user or the Wizard

SvenRoederer, 2018-07-16:
* for me it feels more intuitive, to have the (initial) network configuration in the "network-defaults" than in another extra file. currently there are 3 places to be considered then setting up the network (network-defaults, uplink-package, ffwizard)
* there is no word about removing the current image-flavors with other predefined uplink-types
* if we have found a way to make the uplink-type configurable in the wizard, we can think of eliminating some flavors

bobster-galore, 2018-07-18:
I think it's nice to have a working uplink until u decide to choose something different (like openwrt). 
The only drawback is our "default / intuitive / most used - case" on cpe- or other similar devices, which will be funny.

pmelange, 2018-07-18:
The CPE WAN/LAN port problem is the same either way that Sven decides to implement a default network setup.  

Since I've seen that Sven already has made commits to his branch with this plan, then I don't see the need to discuss or have an RFC.  

A test setup for the tunneldigger solution has alredy been made.  There are a couple issues to work out on the server side, but it seems like converting from notunnel to tunneldigger has little to no impact.  It could be that both uplink types will fit on the images with the possibility to decide in the wizard which to use.

@bobster-galore , I think you meant openvpn and not openwrt.

bobster-galore, 2018-09-13:
I meant openwrt since there is always a working uplink until u decide to use smthg different.

I agree on the cpe-problem, we should develop a nice integration of "I want to share my internet" to make the appropriate settings of the ports.

In general I think we should go for small modular solutions which we can change easily.

bobster-galore, 2018-09-21:
Concerning the CPE-problem I'm working on an "I wan't 2 share my internet" solution https://github.com/freifunk-berlin/firmware-packages/blob/lan-port-on-sharenet/addons/freifunk-berlin-configtools/sharenet-switch.sh where we could switch the port settings accordingly.
What's next?

pmelange, 2018-09-21:
A build with the changes which people can test would be the next step.  Then a PR for someone to review.  Finally a merge.