freifunk-berlin/firmware #580: tunnel-berlin images in "master" and "SAm0815_experimental" not working

pmelange, 2018-07-31:
This is happening on freshly installed and configured (through the wizard) tunnel-berlin images in "master" and "SAm0815_experimental".  This is possibly also an issue in other builds/branches, although I haven't tested them.

* Upon startup, openvpn is not started automatically (noting related to openvpn in the logfile).
* Running "/etc/init.d/openvpn restart" the following happens.
```
~# /etc/init.d/openvpn restart
Command failed: Not found
```
* After running the above command, openvpn _does_ start.  But it doesn't work properly.
```
Tue Jul 31 14:50:11 2018 authpriv.info dropbear[2942]: Child connection from fdd2:4dde:2cc1::d43:40926
Tue Jul 31 14:50:12 2018 kern.notice kernel: [  105.382314] random: crng init done
Tue Jul 31 14:50:14 2018 authpriv.notice dropbear[2942]: Password auth succeeded for 'root' from fdd2:4dde:2cc1::d43:40926
Tue Jul 31 14:51:14 2018 daemon.notice openvpn(ffuplink)[2974]: OpenVPN 2.4.5 mips-openwrt-linux-gnu [SSL (mbed TLS)] [EPOLL] [AEAD]
Tue Jul 31 14:51:14 2018 daemon.notice openvpn(ffuplink)[2974]: library versions: mbed TLS 2.11.0
Tue Jul 31 14:51:14 2018 daemon.warn openvpn(ffuplink)[2974]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Tue Jul 31 14:51:14 2018 daemon.warn openvpn(ffuplink)[2974]: WARNING: failed to personalise random
Tue Jul 31 14:51:14 2018 daemon.warn openvpn(ffuplink)[2974]: ******* WARNING *******: '--cipher none' was specified. This means NO encryption will be performed and tunnelled data WILL be transmitted in clear text over the network! PLEASE DO RECONSIDER THIS SETTING!
Tue Jul 31 14:51:14 2018 daemon.notice openvpn(ffuplink)[2974]: TCP/UDP: Preserving recently used remote address: [AF_INET]217.197.83.193:1194
Tue Jul 31 14:51:14 2018 daemon.notice openvpn(ffuplink)[2974]: UDPv4 link local (bound): [AF_INET][undef]:1194
Tue Jul 31 14:51:14 2018 daemon.notice openvpn(ffuplink)[2974]: UDPv4 link remote: [AF_INET]217.197.83.193:1194
Tue Jul 31 14:51:15 2018 daemon.warn openvpn(ffuplink)[2974]: WARNING: 'link-mtu' is used inconsistently, local='link-mtu 1525', remote='link-mtu 1526'
Tue Jul 31 14:51:15 2018 daemon.warn openvpn(ffuplink)[2974]: WARNING: 'comp-lzo' is present in remote config but missing in local config, remote='comp-lzo'
Tue Jul 31 14:51:15 2018 daemon.notice openvpn(ffuplink)[2974]: [freifunk-gw01.in-berlin.de] Peer Connection Initiated with [AF_INET]217.197.83.193:1194
Tue Jul 31 14:51:16 2018 daemon.notice netifd: Interface 'ffuplink' is enabled
Tue Jul 31 14:51:16 2018 daemon.notice netifd: Network device 'ffuplink' link is up
Tue Jul 31 14:51:16 2018 daemon.notice openvpn(ffuplink)[2974]: TUN/TAP device ffuplink opened
Tue Jul 31 14:51:16 2018 daemon.notice netifd: Interface 'ffuplink' has link connectivity
Tue Jul 31 14:51:16 2018 daemon.notice openvpn(ffuplink)[2974]: do_ifconfig, tt->did_ifconfig_ipv6_setup=0
Tue Jul 31 14:51:16 2018 daemon.notice netifd: Interface 'ffuplink' is setting up now
Tue Jul 31 14:51:16 2018 daemon.notice openvpn(ffuplink)[2974]: /sbin/ifconfig ffuplink 172.31.241.61 netmask 255.255.255.0 mtu 1500 broadcast 172.31.241.255
Tue Jul 31 14:51:16 2018 daemon.notice netifd: Interface 'ffuplink' is now up
Tue Jul 31 14:51:16 2018 daemon.notice openvpn(ffuplink)[2974]: /lib/freifunk/ffvpn-up.sh ffuplink 1500 1552 172.31.241.61 255.255.255.0 init
Tue Jul 31 14:51:16 2018 user.debug up-down-ffvpn: no route_net_gateway env var from openvpn!
Tue Jul 31 14:51:16 2018 daemon.warn openvpn(ffuplink)[2974]: WARNING: this configuration may cache passwords in memory -- use the auth-nocache option to prevent this
Tue Jul 31 14:51:16 2018 daemon.notice openvpn(ffuplink)[2974]: Initialization Sequence Completed
Tue Jul 31 14:51:18 2018 daemon.info modprobe: Usage: 	modprobe [-q] filename
Tue Jul 31 14:51:18 2018 daemon.err modprobe: ifb is already loaded
Tue Jul 31 14:51:18 2018 daemon.err modprobe: cls_u32 is already loaded
Tue Jul 31 14:51:18 2018 daemon.err modprobe: em_u32 is already loaded
Tue Jul 31 14:51:18 2018 daemon.err modprobe: act_connmark is already loaded
Tue Jul 31 14:51:18 2018 daemon.err modprobe: act_mirred is already loaded
Tue Jul 31 14:51:18 2018 daemon.err modprobe: sch_ingress is already loaded
Tue Jul 31 14:51:18 2018 daemon.err modprobe: cls_fw is already loaded
Tue Jul 31 14:51:18 2018 daemon.err modprobe: sch_hfsc is already loaded
Tue Jul 31 14:51:19 2018 user.notice up-down-ffvpn: ugw: 192.168.1.1 dev: ffuplink remote: 255.255.255.0 gw: 172.31.241.1 src: 172.31.241.61 mask: 255.255.255.0
Tue Jul 31 14:51:19 2018 user.notice firewall: Reloading firewall due to ifup of ffuplink (ffuplink)
Tue Jul 31 14:51:19 2018 daemon.info olsrd_hotplug: [OK] ifup: 'ffuplink' => 'ffuplink'
Tue Jul 31 14:51:19 2018 daemon.debug olsrd_hotplug: [OK] interface 'ffuplink' => 'ffuplink' not used for olsrd
Tue Jul 31 14:51:20 2018 daemon.info olsrd_hotplug: [OK] ifup: 'ffuplink' => 'ffuplink'
Tue Jul 31 14:51:20 2018 daemon.debug olsrd_hotplug: [OK] interface 'ffuplink' => 'ffuplink' not used for olsrd6
Tue Jul 31 14:51:20 2018 user.notice ff-userlog: ffuplink interface is up
Tue Jul 31 14:51:20 2018 user.notice ff-userlog: creating ffuplink ip-rules
Tue Jul 31 14:51:20 2018 user.notice ff-userlog: UCI did not return a valid IP-net for ffuplink; querying directly with ip-tool
Tue Jul 31 14:51:20 2018 user.notice ff-userlog: ffuplink-interface is setup
Tue Jul 31 14:51:26 2018 daemon.err openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)
Tue Jul 31 14:51:29 2018 daemon.err odhcp6c[1414]: Failed to send DHCPV6 message to ff02::1:2 (Permission denied)
Tue Jul 31 14:51:30 2018 daemon.warn odhcpd[994]: DHCPV6 RENEW IA_NA from 00041255ae230e49244278313f53c3a1f05f on br-dhcp: ok fdd2:4dde:2cc1::d43/128
Tue Jul 31 14:51:36 2018 daemon.err openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)
Tue Jul 31 14:51:46 2018 daemon.err openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)
Tue Jul 31 14:51:56 2018 daemon.err openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)
Tue Jul 31 14:52:07 2018 daemon.err openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)
Tue Jul 31 14:52:17 2018 daemon.err openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)
Tue Jul 31 14:52:28 2018 daemon.err openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)
Tue Jul 31 14:52:38 2018 daemon.err openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)
Tue Jul 31 14:52:48 2018 daemon.err openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)
Tue Jul 31 14:52:58 2018 daemon.err openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)
Tue Jul 31 14:53:08 2018 daemon.err openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)
Tue Jul 31 14:53:42 2018 daemon.err odhcp6c[1414]: Failed to send DHCPV6 message to ff02::1:2 (Permission denied)
Tue Jul 31 14:54:00 2018 daemon.warn odhcpd[994]: DHCPV6 RENEW IA_NA from 00041255ae230e49244278313f53c3a1f05f on br-dhcp: ok fdd2:4dde:2cc1::d43/128
[... 27 lines cut ...]
Tue Jul 31 14:54:17 2018 daemon.err modprobe: ifb is already loaded
Tue Jul 31 14:54:17 2018 daemon.err modprobe: cls_u32 is already loaded
Tue Jul 31 14:54:17 2018 daemon.err modprobe: em_u32 is already loaded
Tue Jul 31 14:54:17 2018 daemon.err modprobe: act_connmark is already loaded
Tue Jul 31 14:54:17 2018 daemon.err modprobe: act_mirred is already loaded
Tue Jul 31 14:54:17 2018 daemon.err modprobe: sch_ingress is already loaded
Tue Jul 31 14:54:17 2018 daemon.err modprobe: cls_fw is already loaded
Tue Jul 31 14:54:17 2018 daemon.err modprobe: sch_hfsc is already loaded
Tue Jul 31 14:54:17 2018 user.notice up-down-ffvpn: ugw: 192.168.1.1 dev: ffuplink remote: 255.255.255.0 gw: 172.31.241.1 src: 172.31.241.61 mask: 255.255.255.0
Tue Jul 31 14:54:18 2018 user.notice firewall: Reloading firewall due to ifup of ffuplink (ffuplink)
Tue Jul 31 14:54:18 2018 daemon.info olsrd_hotplug: [OK] ifup: 'ffuplink' => 'ffuplink'
Tue Jul 31 14:54:18 2018 daemon.debug olsrd_hotplug: [OK] interface 'ffuplink' => 'ffuplink' not used for olsrd
Tue Jul 31 14:54:18 2018 daemon.info olsrd_hotplug: [OK] ifup: 'ffuplink' => 'ffuplink'
Tue Jul 31 14:54:18 2018 daemon.debug olsrd_hotplug: [OK] interface 'ffuplink' => 'ffuplink' not used for olsrd6
Tue Jul 31 14:54:18 2018 user.notice ff-userlog: ffuplink interface is up
[... cut ...]
Tue Jul 31 14:54:19 2018 user.notice ff-userlog: ffuplink-interface is setup
Tue Jul 31 14:54:25 2018 daemon.err openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)
```
* The only difference in the options in /etc/config/openvpn between Hedy-1.0.1 and the master/SAm branches is that in Hedy-1.0.1 there is the following option
```
option local '192.168.1.4'
```
* After adding the "local" option, there is no difference.  Openvpn does not start automatically and manually starting has the same results as shown above.

pmelange, 2018-08-01:
Update:  I have found one of the problems... The hotplug script in firmware-packages/defaults/freifunk-berlin-openvpn-files/openvpn/60-ffopenvpn declared variables with the "local" declaration.  I have made commit a87f7e53259e1f588f7747a16557c895a3fe7ce8 with the needed changes.

Now OpenVPN starts at boot time and there is not longer an error when running "/etc/init.d/openvpn restart".  That, at least, is step 1 of the problem.

But the rest of the problems from the description persist.

If it helps, there are errors in /tmp/ffvpn-up.log 
```
net.ipv6.conf.ffuplink.disable_ipv6 = 1
RTNETLINK answers: No such process
RTNETLINK answers: No such process
```

SvenRoederer, 2018-08-02:
I think the problem comes from here:

> Tue Jul 31 14:51:15 2018 daemon.warn openvpn(ffuplink)[2974]: WARNING: 'link-mtu' is used inconsistently, local='link-mtu 1525', remote='link-mtu 1526'
Tue Jul 31 14:51:15 2018 daemon.warn openvpn(ffuplink)[2974]: WARNING: 'comp-lzo' is present in remote config but missing in local config, remote='comp-lzo'

and a result is (check a search-engine for openvpn lzo "Invalid argument (code=22)")

> openvpn(ffuplink)[2974]: write to TUN/TAP : Invalid argument (code=22)

probably related to some deprecation of the comp-lzo setting in openvpn

have you tried to build a version w/o the mbedtls-1024 patch? Did that work?

pmelange, 2018-08-02:
I have built the master branch without the 1024 patch with the same results.  I also removed vpnbypass with and without the 1024 patch.  Same results.

Here is the logfile.  It doesn't look different from before
```
Thu Aug  2 22:46:39 2018 user.notice ff-vpn-hotplug: Starting OpenVPN on WAN interface
Thu Aug  2 22:46:40 2018 daemon.notice openvpn(ffuplink)[3070]: OpenVPN 2.4.5 mips-openwrt-linux-gnu [SSL (mbed TLS)] [EPOLL] [AEAD]
Thu Aug  2 22:46:40 2018 daemon.notice openvpn(ffuplink)[3070]: library versions: mbed TLS 2.8.0
Thu Aug  2 22:46:40 2018 daemon.warn openvpn(ffuplink)[3070]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Thu Aug  2 22:46:40 2018 daemon.warn openvpn(ffuplink)[3070]: WARNING: failed to personalise random
Thu Aug  2 22:46:40 2018 daemon.warn openvpn(ffuplink)[3070]: ******* WARNING *******: '--cipher none' was specified. This means NO encryption will be performed and tunnelled data WILL be transmitted in clear text over the network! PLEASE DO RECONSIDER THIS SETTING!
Thu Aug  2 22:46:40 2018 daemon.notice openvpn(ffuplink)[3070]: TCP/UDP: Preserving recently used remote address: [AF_INET]46.4.104.30:1194
Thu Aug  2 22:46:40 2018 daemon.notice openvpn(ffuplink)[3070]: UDPv4 link local (bound): [AF_INET]192.168.200.3:1194
Thu Aug  2 22:46:40 2018 daemon.notice openvpn(ffuplink)[3070]: UDPv4 link remote: [AF_INET]46.4.104.30:1194
Thu Aug  2 22:46:41 2018 daemon.err odhcp6c[1449]: Failed to send DHCPV6 message to ff02::1:2 (Permission denied)
Thu Aug  2 22:46:43 2018 kern.notice kernel: [   64.833292] random: crng init done
Thu Aug  2 22:47:00 2018 cron.err crond[2977]: time disparity of 1681 minutes detected
Thu Aug  2 22:47:18 2018 daemon.err odhcp6c[1449]: Failed to send DHCPV6 message to ff02::1:2 (Permission denied)
Thu Aug  2 22:47:40 2018 daemon.notice openvpn(ffuplink)[3070]: [UNDEF] Inactivity timeout (--ping-restart), restarting
Thu Aug  2 22:47:40 2018 daemon.notice openvpn(ffuplink)[3070]: SIGUSR1[soft,ping-restart] received, process restarting
Thu Aug  2 22:47:45 2018 daemon.warn openvpn(ffuplink)[3070]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Thu Aug  2 22:47:45 2018 daemon.notice openvpn(ffuplink)[3070]: TCP/UDP: Preserving recently used remote address: [AF_INET]217.197.83.193:1194
Thu Aug  2 22:47:45 2018 daemon.notice openvpn(ffuplink)[3070]: UDPv4 link local (bound): [AF_INET]192.168.200.3:1194
Thu Aug  2 22:47:45 2018 daemon.notice openvpn(ffuplink)[3070]: UDPv4 link remote: [AF_INET]217.197.83.193:1194
Thu Aug  2 22:47:47 2018 daemon.warn openvpn(ffuplink)[3070]: WARNING: 'link-mtu' is used inconsistently, local='link-mtu 1525', remote='link-mtu 1526'
Thu Aug  2 22:47:47 2018 daemon.warn openvpn(ffuplink)[3070]: WARNING: 'comp-lzo' is present in remote config but missing in local config, remote='comp-lzo'
Thu Aug  2 22:47:47 2018 daemon.notice openvpn(ffuplink)[3070]: [freifunk-gw01.in-berlin.de] Peer Connection Initiated with [AF_INET]217.197.83.193:1194
Thu Aug  2 22:47:48 2018 daemon.notice netifd: Interface 'ffuplink' is enabled
Thu Aug  2 22:47:48 2018 daemon.notice netifd: Network device 'ffuplink' link is up
Thu Aug  2 22:47:48 2018 daemon.notice openvpn(ffuplink)[3070]: TUN/TAP device ffuplink opened
Thu Aug  2 22:47:48 2018 daemon.notice netifd: Interface 'ffuplink' has link connectivity
Thu Aug  2 22:47:48 2018 daemon.notice openvpn(ffuplink)[3070]: do_ifconfig, tt->did_ifconfig_ipv6_setup=0
Thu Aug  2 22:47:48 2018 daemon.notice netifd: Interface 'ffuplink' is setting up now
Thu Aug  2 22:47:48 2018 daemon.notice openvpn(ffuplink)[3070]: /sbin/ifconfig ffuplink 172.31.241.61 netmask 255.255.255.0 mtu 1500 broadcast 172.31.241.255
Thu Aug  2 22:47:48 2018 daemon.notice netifd: Interface 'ffuplink' is now up
Thu Aug  2 22:47:48 2018 daemon.notice openvpn(ffuplink)[3070]: /lib/freifunk/ffvpn-up.sh ffuplink 1500 1552 172.31.241.61 255.255.255.0 init
Thu Aug  2 22:47:48 2018 user.debug up-down-ffvpn: no route_net_gateway env var from openvpn!
Thu Aug  2 22:47:48 2018 daemon.warn openvpn(ffuplink)[3070]: WARNING: this configuration may cache passwords in memory -- use the auth-nocache option to prevent this
Thu Aug  2 22:47:48 2018 daemon.notice openvpn(ffuplink)[3070]: Initialization Sequence Completed
Thu Aug  2 22:47:50 2018 daemon.info modprobe: Usage: 	modprobe [-q] filename
Thu Aug  2 22:47:50 2018 daemon.err modprobe: ifb is already loaded
Thu Aug  2 22:47:50 2018 daemon.err modprobe: cls_u32 is already loaded
Thu Aug  2 22:47:50 2018 daemon.err modprobe: em_u32 is already loaded
Thu Aug  2 22:47:50 2018 daemon.err modprobe: act_connmark is already loaded
Thu Aug  2 22:47:50 2018 daemon.err modprobe: act_mirred is already loaded
Thu Aug  2 22:47:50 2018 daemon.err modprobe: sch_ingress is already loaded
Thu Aug  2 22:47:50 2018 daemon.err modprobe: cls_fw is already loaded
Thu Aug  2 22:47:50 2018 user.notice up-down-ffvpn: ugw: 192.168.200.1 dev: ffuplink remote: 255.255.255.0 gw: 172.31.241.1 src: 172.31.241.61 mask: 255.255.255.0
Thu Aug  2 22:47:50 2018 daemon.err modprobe: sch_hfsc is already loaded
Thu Aug  2 22:47:51 2018 user.notice firewall: Reloading firewall due to ifup of ffuplink (ffuplink)
Thu Aug  2 22:47:51 2018 daemon.info olsrd_hotplug: [OK] ifup: 'ffuplink' => 'ffuplink'
Thu Aug  2 22:47:51 2018 daemon.debug olsrd_hotplug: [OK] interface 'ffuplink' => 'ffuplink' not used for olsrd
[... cut ...]
22:50:59.066179 IP (tos 0x0, ttl 64, id 3941, offset 0, flags [DF], proto UDP (17), length 136)
    192.168.200.3.1194 > 217.197.83.193.1194: [bad udp cksum 0xb6b8 -> 0x3675!] UDP, length 108
22:51:00.074120 IP (tos 0x0, ttl 64, id 3973, offset 0, flags [DF], proto UDP (17), length 136)
    192.168.200.3.1194 > 217.197.83.193.1194: [bad udp cksum 0xb6b8 -> 0x8ba0!] UDP, length 108
22:51:01.082085 IP (tos 0x0, ttl 64, id 4039, offset 0, flags [DF], proto UDP (17), length 136)
    192.168.200.3.1194 > 217.197.83.193.1194: [bad udp cksum 0xb6b8 -> 0x0392!] UDP, length 108
```

sarumpaet, 2018-08-05:
Perhaps https://github.com/freifunk-berlin/firmware/commit/087680a38798ca32a7c96c2932cbd516638f0251 ?
Test with the standard OpenWrt OpenVPN package?

pmelange, 2018-08-05:
I'm building now with these changes removed, but with the 1024 patch and vpnbypass in.  I'll let you know tomorrow.

pmelange, 2018-08-06:
With the changes in 087680a removed, I still have the same problems.

booo, 2018-08-06:
Please check against one of the VPN03 servers.

pmelange, 2018-08-06:
With the changes in 087680a removed, and with the 1024 patch and vpnbypass in, configured to use vpn03 (with the correct certs, keys, ...), I still have the same problems.

pmelange, 2018-08-06:
I just tried out openwrt 18.06.0 with openvpn (mbedtls).  I'm also getting these "code=22" errors.

pmelange, 2018-08-06:
With 17.05.1 I don't get these errors.

pmelange, 2018-08-06:
https://community.openvpn.net/openvpn/ticket/952 might explain what is going on.

In Hedy 1.0.1 we use openvpn 2.4.4
In master we use openvpn 2.4.5
in 17.01.5 they use openvpn 2.4.4
in 18.06.0 they use openvpn 2.4.5

On server vpn03f.berlin.freifunk.net we use 2.3.12 
On server vm02.roellig-it.de) we use 2.3.10

It would be really great if someone with more experience with openvpn took a look at this.

SvenRoederer, 2018-08-06:
freifunk-gw01.in-berlin.de : Version: 2.4.0-6+deb9u2

pmelange, 2018-08-10:
I have created a ticket with openwrt:
https://bugs.openwrt.org/index.php?do=details&task_id=1762

@SvenRoederer, could you post the server config for freifunk-gw01.in-berlin.de?

pmelange, 2018-08-10:
@lynxis As per your request, I have posted a ticket with openwrt.
https://bugs.openwrt.org/index.php?do=details&task_id=1762

pmelange, 2018-08-20:
Sorry, I wrote too soon.  The ```option compress lzo``` works with 18.06.01 but not with the freifunk firmware.

SvenRoederer, 2018-08-20:
As you can see from the sever-configs, we want no compression at all. So compression of data on the client side but no decompression on the server-side will fail logically.

according to the openvpn-ticket referenced above, one solution would be to enable lzo on both sides of the tunnel. but this requires a change on all clients.

pmelange, 2018-08-21:
Having compression on the client->server and no compression on the server->client (and visa-versa) is a totally valid configuration.  The reason that it didn't work with the freifunk firmware, like I wrote above, is because of 087680a. 

It is also possible to have the server push no compression to the clients after they connect.

The thing that is a problem is that I still can't figure out a way to have the clients do no compression.  Simply leaving out the compress option will not work.  I have tried various versions of the compress option, but I still haven't found one that works.

So, my suggestion is that we re-enable the ```CONFIG_OPENVPN_mbedtls_ENABLE_LZO``` compile option and have the server push no compression.  The downside is a bigger binary and dependency on liblzo.

This will be my last post until some time early Sept.  If someone else wants to take over this issue, that would be great.

https://community.openvpn.net/openvpn/wiki/DeprecatedOptions
https://community.openvpn.net/openvpn/ticket/952
https://wiki.openwrt.org/inbox/vpn.howto

pmelange, 2018-09-10:
I hacked the openvpn init script to allow for ```option compress '1'``` or ```option compress '0'```.  The former adds a line with ```compress``` in /var/etc/openvpn-ffuplink.conf.  This should be what we want.  The connection to the server gets established, but I am not able to send any data through.  Evenutally the connection times-out and reestablishes.

See https://bugs.openwrt.org/index.php?do=details&task_id=1762

I'm not sure what to do next.  Maybe we need to explicitly put ```compress``` into the server config file.  I'm not sure how this will impact our current clients.

Anybody have any other ideas?
Should we just reenable compiling with liblzo support (and add ```option compress 'lzo'```), since this method works?

SvenRoederer, 2018-09-12:
As mentioned in my last post (https://github.com/freifunk-berlin/firmware/issues/580#issuecomment-414725804) I had no luck with using the original 18.06-openwrt packahe of openvpn. So I don't believe in enabling lzo-support will fix this.
But feel free to prove me wrong by creating a branch with enabled liblzo.

bobster-galore, 2018-09-13:
What is the problem?
funny options or different versions?

pmelange, 2018-09-13:
I have created branch openvpn245.  I have also installed it on my rb750gr3.

@SvenRoederer , @bobster-galore , @booo I have added your ssh keys to the router.  It is reachable via the backbone.  The host is ptest3.olsr

I will be gone until Sat evening.  Until then, have fun.  And if anyone wants to merge the changes on the openvpn245 branches in firmware and firmware-packages, then that would be really cool.

pmelange, 2018-09-13:
Oh, and I haven't done migration yet. I can do it on Sunday.

pmelange, 2018-09-16:
There are now two PR's ready for someone to take a look at and (hopefully) merge.

I saw that @SvenRoederer pushed a commit to the master branch to reeable compiling with liblzo support.  I believe the PR that I created is complete and with better comments.  Also, I find it better to have the compile option commented out (readability and understandability) then just simply deleting the compile options.  

I would recommend deleting that commit and merging the openvpn245 branch.  I

pmelange, 2018-09-18:
The system has been running for 5 days now.  @SvenRoederer @booo @bobster-galore is there any reason to keep the box running?  If nobody objects, I would unplug the router tomorrow evening.

bobster-galore, 2018-09-19:
I had no idea what to test. Let it go unplugged!

bobster-galore, 2018-09-21:
How do we fix this for 1.1.0?

pmelange, 2018-09-21:
There are 2 PR's ready to fix this issue.  They need a review and hopefully to be merged.

keks1337, 2018-09-30:
have same issues. use last build.

`Sun Sep 30 17:52:32 2018 kern.notice kernel: [    0.000000] Linux version 4.14.54 (pfleger@buildbot) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r6907+293-7e15e21)) #0 Tue Jul 31 12:06:07 2018
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Board has DDR2
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Analog PMU set to hw control
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Digital PMU set to hw control
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7628AN ver:1 eco:2
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] MIPS: machine is TP-Link TL-WR841N v13
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Determined physical RAM map:
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000]  memory: 04000000 @ 00000000 (usable)
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sun Sep 30 17:52:32 2018 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Sun Sep 30 17:52:32 2018 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Zone ranges:
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Movable zone start for each node
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Early memory node ranges
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Sun Sep 30 17:52:32 2018 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Sun Sep 30 17:52:32 2018 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 803c8b30, node_mem_map 81000040
Sun Sep 30 17:52:32 2018 kern.debug kernel: [    0.000000]   Normal zone: 128 pages used for memmap
Sun Sep 30 17:52:32 2018 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sun Sep 30 17:52:32 2018 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Sun Sep 30 17:52:32 2018 kern.notice kernel: [    0.000000] random: get_random_bytes called from 0x803cc72c with crng_init=0
Sun Sep 30 17:52:32 2018 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Sun Sep 30 17:52:32 2018 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
Sun Sep 30 17:52:32 2018 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Writing ErrCtl register=00073350
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Readback ErrCtl register=00073350
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] Memory: 60568K/65536K available (3345K kernel code, 170K rwdata, 364K rodata, 208K init, 208K bss, 4968K reserved, 0K cma-reserved)
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] NR_IRQS: 256
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] intc: using register map from devicetree
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] CPU Clock: 580MHz
Sun Sep 30 17:52:32 2018 kern.crit kernel: [    0.000000] timer_probe: no matching timers found
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.000012] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.007535] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.073498] pid_max: default: 32768 minimum: 301
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.078187] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.084537] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.096990] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.106479] futex hash table entries: 256 (order: -1, 3072 bytes)
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.112432] pinctrl core: initialized pinctrl subsystem
Sun Sep 30 17:52:32 2018 kern.info kernel: [    0.118008] NET: Registered protocol family 16
[... cut ...]
Sun Sep 30 17:55:24 2018 daemon.err openvpn(ffuplink)[2638]: write to TUN/TAP : Invalid argument (code=22)
Sun Sep 30 17:55:35 2018 daemon.err openvpn(ffuplink)[2638]: write to TUN/TAP : Invalid argument (code=22)
Sun Sep 30 17:55:40 2018 daemon.warn odhcpd[913]: DHCPV6 RENEW IA_NA from 000100012037e79fc0143dcc2a0d on br-dhcp: ok fdd7:2a6c:57a0::840/128
Sun Sep 30 17:56:13 2018 daemon.err uhttpd[2035]: luci: accepted login on /admin for root from fdd7:2a6c:57a0::29f5:25bc:484a:29c3
`

some ideas?

pmelange, 2018-12-12:
A new commit has been added to openwrt-18-06
https://github.com/openwrt/openwrt/commit/d5afaa41148b363128626135bfb73b156887cddb
This may solve this issue

pmelange, 2018-12-12:
Even the description here has changed...
https://community.openvpn.net/openvpn/wiki/DeprecatedOptions#a--comp-lzo

```Contrary to prior statements --comp-lzo no is not compatible with the --compress counterpart. Therefore openvpn needs to keep supporting --comp-lzo no for backward compatibility. ```

SvenRoederer, 2018-12-12:
in the OpenVPN-wiki the change of the removal is dated to 20.Oct 2018 (https://community.openvpn.net/openvpn/wiki/DeprecatedOptions?sfp_email=&sfph_mail=&action=diff&version=13&old_version=11&sfp_email=&sfph_mail=)

pmelange, 2018-12-13:
The lates commit to 2.4.5 is "Latest commit 27a2e01 on Feb 28".  Since 18.06 is still using 2.4.5 then we might be in luck.

pmelange, 2018-12-13:
My recommendation is the following:
* set up an openvpn service with the new compression option on another port number
* as of Hedy 1.1.x we change the openvpn options via mirgration script to the new port and config

We can also set up the service like this on all the remaining vpn03 servers.

SvenRoederer, 2018-12-14:
this I had done on the in-berlin gateway once

pmelange, 2018-12-21:
I have built locally and tested this issue with commit https://github.com/openwrt/openwrt/commit/d5afaa41148b363128626135bfb73b156887cddb integrated.  Now OpenVpn 2.4.5 works.

Shall we close this issue?  If we want to create a new setup on the server side and client side for 1.1.x or possibly 1.2.x, then I think it should go in a new issue.

SvenRoederer, 2018-12-29:
let's close this here. changes to the tunnel-setup should go into the tunnel-repo or better on the mailinglist

closed 2018-12-29 by SvenRoederer