freifunk-berlin/firmware #584: building of owm-app should honor the CONFIG_LUCI_SRCDIET flag

SvenRoederer, 2018-08-20:
luci-app-owm is always build via precompiling with luac.

According to https://www.lua.org/manual/5.1/luac.html the only advantage is faster loading. Is this the reason for the current way it is build? 

For development build it would be more feasible to have the original source on the router.

booo, 2018-08-21:
Let us disable the precompile option.

bobster-galore, 2018-09-13:
Yes, disable it! 
Probably not only for development but also for production.
I couldn't find any speed advantages until now and what makes me wonder (from s.a.):
 ```
main advantages of precompiling chunks are: faster loading
Precompiling does not imply faster execution
 ```
The advantages sound to me very academic at least in freifunk context.