freifunk-berlin/firmware #585: do not publish MAC-adresses on OWM

SvenRoederer, 2018-08-27:
As mentioned in https://github.com/freifunk-berlin/firmware/issues/507#issuecomment-414404087 I wonder if we should drop submitting the anonymized MAC-addresses from the owm-script.

Do we need to collect and publish this data?
For what purpose do we collect them?

bobster-galore, 2018-08-28:
I think due to data avoidance and data economy we could drop MAC-addresses in owm. I see no use at all in context of freifunk. 
If there is a need to know MAC-addresses it is only on the respective node, where it can be easily obtained.
I'm even not certain that a number of clients is desirable, since we don't have this type of record-of-clients-on-a-node maps. So we don't need to know.

SvenRoederer, 2018-08-28:
for hopglass the # of clients is just disabled by setting it to "0" (https://github.com/freifunk-berlin/hopglass.berlin.freifunk.net/blob/master/owm2ffmap/owm2ffmap.py#L196). So if we want tp present the clients-number we just can do.
But even then, we do not need the clients-MAC.

pmelange, 2018-09-05:
I support the plan do get rid of the MAC addresses.  Having the number of active users is, on the other hand, potentially valuable/interesting.

bobster-galore, 2018-09-13:
The number of clients is also available at the node, no need to know it online. We shouldn't offer data for collectors.

bobster-galore, 2018-09-21:
We discussed that on our Spandau meeting and were all (4) in favour of dropping mac address as well as number of clients in owm.