freifunk-berlin/firmware #587: SAm0815-experimental: no entry in /etc/config/dhcp

pmelange, 2018-09-18:
With 2966a76 from SAM0815-Exterimental there is a problem with DHCP on the client network.  There is no entry in /etc/config/dhcp for the 'dhcp' network.

Manually adding the entry (and also probably by clicking) solves the problem.

everloop2, 2018-09-28:
Same here.

Last image with no quirks after FFwizard from 22.07.2018 (alpha-8d54918).

Few days later, 28.07.2018 (alpha-2b09271): no dhcp server was set up.

Workaround: give urself static IP and connect dhcp.

Workaround2: backup configuration and restore after flashing or try keep settings when flashing firmware image.

SvenRoederer, 2018-10-18:
The problem seems to come from here: https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/luci-app-ffwizard-berlin/luasrc/model/cbi/freifunk/assistent/wireless.lua#L242

```
--DHCP CONFIG bridge for wifi APs
  local dhcpbase = uci:get_all("freifunk", "dhcp") or {}
  util.update(dhcpbase, uci:get_all(community, "dhcp") or {})
  dhcpbase.interface = "dhcp"
```
when removing the `uci:get_all` and using an empty `dhcpbase` setting up the DHCP-server section works again.

As this code was not changed on our side, I assume a change upstream.

SvenRoederer, 2018-10-20:
https://github.com/freifunk-berlin/firmware-packages/commit/1f6ecc0cb43cf5d9324d710792afe7b5af1c3dad is a temporarily fix

SvenRoederer, 2018-10-27:
I integrated a workaround into branch https://github.com/freifunk-berlin/firmware-packages/tree/80211s-mesh