freifunk-berlin/firmware #588: SAm0815-experimental: wifi not enabled

pmelange, 2018-09-18:
On 2966a76 from SAm0815-experimental, the wireless setting are left at disabled after running the wizard.

Also, there is no mesh/adhoc interface.

everloop2, 2018-09-28:
Same here.

Last image with no quirks after FFwizard from 22.07.2018 (alpha-8d54918).

Few days later, 28.07.2018 (alpha-2b09271): no wireless0 (mesh interface) and no dhcp server set up.

Workaround: backup configuration and restore after flashing or try keep settings when flashing firmware image.

SvenRoederer, 2018-10-20:
https://github.com/freifunk-berlin/firmware-packages/commit/1f6ecc0cb43cf5d9324d710792afe7b5af1c3dad is a temporarily fix  

it's the same issue as https://github.com/freifunk-berlin/firmware/issues/587

SvenRoederer, 2018-10-27:
I integrated a workaround into branch https://github.com/freifunk-berlin/firmware-packages/tree/80211s-mesh