freifunk-berlin/firmware #591: Integrate PoE Passthrough into uci for nanostations and cpe

bobster-galore, 2018-09-22:
It would be nice to steer PoE Passthrough thru uci instead of fumbling with files. How can we cherrypickpocket from here https://github.com/freifunk-gluon/gluon/pull/651/commits/d202964048135b3a78eb3190704938e8b8231e02 ?

SvenRoederer, 2018-10-12:
my CPE210 (Hedy1.0.1) has the following section in config/system

> config gpio_switch 'poe_passthrough'
        option name 'PoE Passthrough'
        option gpio_pin '20'
        option value '0'

and for an ER-x I've seen it was working

bobster-galore, 2018-10-26:
Ok, with
```
config gpio_switch 'poe_passthrough'
	option name 'PoE Passthrough'
	option gpio_pin '20'
	option value '1'
```
it is working. Before there was a typo in the manual entry.

This entry does not come on our migration path. Perhaps it would be nice to integrate it there with "option value '0'" or try to read if there is an poe-entry in rc.local and guess a value of '1'.

bobster-galore, 2018-10-27:
If we want to run only /etc/board.d/03_gpio_switches it is not enough to run with hardcoded board name since the "$CFG" is missing which leads finally to an error *mv file to 'nothing'".

Perhaps we could selctively run [/bin/board_detect ](https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/base-files/files/bin/board_detect;hb=39e5e17045aceb2bfbd6b5c6ecfd6cfbce2f3311) only for 03_gpio_switches because it delivers the "$CFG".

bobster-galore, 2018-10-27:
Both routers do have the right entries in /etc/board.json, so it's not missing in general but not written to /etc/config/system.

bobster-galore, 2018-10-27:
As [tweak mentions](https://openwrt.org/toh/bt/homehub_v5a#example_boardjson_tweak) /etc/board.json only describes hardware whereas [config_generate](https://github.com/openwrt/luci/issues/779#issuecomment-240624571) should alter settings, which it doesn't in my case, unfortunately.
I think the right way is to use config_generate.

pmelange, 2018-10-27:
I have always manually entered the gpio setting in /etc/config/system, following instructions on the openWrt wiki pages.

If the goal is to have an option to turn on/off poe passthrough either through Luci and/or the wizard, then I would suggest having a simple warning next to the option stating that if poe passthrough has been manually added to some script (such as /etc/rc.local) then the user should take responsibility to remove it.  Anyhow, I find that an edge case.

SvenRoederer, 2021-06-13:
During the lifetime of this issue OpenWrt made UCI-setting for GPIO, (there might be also a LuCI-app around). So the issue left is that upgrading from pre-Lede-17.01 the missing UCI-section is not created, which should be a separate issue.

closed 2021-06-13 by SvenRoederer