freifunk-berlin/firmware #594: The OLSR services menu is 404 Not Found

pmelange, 2018-10-14:
Tested with Freifunk Berlin 1.1.0-alpha 4155686 (tunnel-berlin) on a WR842v3

The "Services" menu option on the top of the web interface is not there.  It should be between Statistics and Map.  

Manually entering the address http://frei.funk/cgi-bin/luci/freifunk/services give a 404 Not Found.

pmelange, 2018-10-14:
The nameservice plugin is active and the contents of /var/etc/services.olsr looks to be correct.

mmouselli, 2018-10-21:
The problem has been fixed, you need to update Freifunk's repositories.
https://github.com/openwrt/luci/pull/2238

booo, 2018-10-24:
Fixed with merge https://github.com/openwrt/luci/commit/290610798a4d6ced5d0db31e1cbcad7d758e020f

Once we update the luci feed this should be fixed.

SvenRoederer, 2018-10-24:
@booo but have in mind, that this is only in the master-branch for now

booo, 2018-10-25:
The olsrd library version numbers are everywhere. I created a pull request to get rid of every version number in the code.

closed 2018-11-03 by booo