freifunk-berlin/firmware #603: no-tunnel: when not sharing internet, disable dhcp on ffuplink

pmelange, 2018-10-20:
When a node is setup with ""Freifunk teilnehmen", the ffuplink is still connected to br-wan and makes dhcp requests.  This leads to many warning in the logfile.

pmelange, 2018-10-24:
@SvenRoederer, you have a lot more experience with the wizard and how the routers are set up.  How could the ffuplink interface be set to proto 'none' if sharenet=0?  I've spent a lot of time looking though the code, but I haven't found a sonution yet.  Any help or pointers would be greatly appreciated.

pmelange, 2018-10-24:
After thinking about it more, does it make more sense to set the option ```disabled '1'``` for the ffuplink interface when sharenet=0?

SvenRoederer, 2018-10-24:
it's probably the best to disable the interface by default when it's created. Then it can be enabled on demand

pmelange, 2018-10-25:
That is also possible.  How would you do it?  In the Wizard?  With a hotplug script which checks sharenet?  Wouldn't the hotplug script then have to change /etc/config/network

If it's done in the wizard, where would the code go?

If it's done with the hotplug script, which also changes the /etc/config/network, then I find it problematic. Any commented out sections, after the uci commit, would be lost.   I always find it annoying when my comments in a config file get wiped away magically.

pmelange, 2018-10-29:
@SvenRoederer I see that you have started working on this.  I am willing to help.  Could we make a new feature branch for us to work on and try to define who works on what part?

pmelange, 2018-10-29:
This issue is related to https://github.com/freifunk-berlin/firmware/issues/601

SvenRoederer, 2018-11-19:
When the idea of disabling the ffuplink-interface by default (for any kind of uplink) is working, we just need an additional line in the wizard.

pmelange, 2018-11-19:
tunneldigger is ready.  proto dhcp (no-tunnel) works out of the box.  And openvpn isn't able to start before the wizard.  Cool enhancement.  It also could solve #624 (for the time before the wizard has been ran).