freifunk-berlin/firmware #605: OLSR library plugin version numbers are still referenced in various files

pmelange, 2018-10-21:
These files contain references to OLSR library plugins with the version number.  This need to either be updated to not have a version number (preferred) or to the current version number.

utils/luci-mod-freifunk-ui/luasrc/view/freifunk-map/map.htm
utils/luci-mod-freifunk-ui/luasrc/view/freifunk-map/frame.htm
utils/luci-app-ffwizard-berlin/luasrc/tools/freifunk/assistent/olsr.lua

booo, 2018-10-24:
Can you create a pull request for this from your branch. Thanks a lot.

pmelange, 2018-10-24:
I haven't gotten around to testing it yet.  Maybe in the next couple of days.

pmelange, 2018-10-24:
I just did ```rgrep olsr_ *``` in http://github-com/openwrt/luci and the version numbers are referenced in a lot of places still.  Should we do the work ourselves or add an issue for luci and hope someone gets around to doing and testing the changes?

booo, 2018-11-03:
This was fixed upstream with a merge to the master branch: https://github.com/openwrt/luci/pull/2250.

closed 2018-11-03 by booo