freifunk-berlin/firmware #609: Patch 200-wifi_nolegacy_rates_by_default.patch fails with luci master

pmelange, 2018-10-24:
I just tried doing a build to test another issue which requires the latest from luci.  Unfortunately the build failed at patching 200-wifi_nolegacy_rates_by_default.patch

The feed I used is https://github.com/openwrt/luci.git^290610798a4d6ced5d0db31e1cbcad7d758e020f

```
Applying patch 200-wifi_nolegacy_rates_by_default.patch
can't find file to patch at input line 5
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/wifi.lua b/modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/wifi.lua
|index c0bb38030..877395a36 100644
|--- a/feeds/luci/modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/wifi.lua
|+++ b/feeds/luci/modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/wifi.lua
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch 200-wifi_nolegacy_rates_by_default.patch does not apply (enforce with -f)
Makefile:92: recipe for target '.stamp-patched' failed
make: *** [.stamp-patched] Error 1
```

SvenRoederer, 2018-10-24:
please check https://github.com/freifunk-berlin/firmware/commit/02a49c7991c6c502a3f5a0954daf4819b1e1f74a

pmelange, 2018-10-25:
Thanks for the pointer.  I cherry-picked the commit into my working branch, updated the feed, and am building locally.  Hopefully I'll be able to test later today.

pmelange, 2018-11-04:
This will be resolved when #615 gets merged

SvenRoederer, 2018-11-07:
btw. is this also relevant for 18.06 branch?

SvenRoederer, 2019-01-30:
can also be closed as of https://github.com/freifunk-berlin/firmware/pull/615#issuecomment-459145262

closed 2019-01-30 by SvenRoederer