freifunk-berlin/firmware #613: luci-master: after logging in, the user is directed to the changePassword page

pmelange, 2018-11-03:
The Luci dispatcher no longer forwards the user to the "Freifunk Overview" page but instead prompts the user to set the root password.

pmelange, 2018-11-18:
I have made changes to the wizard and mod-freifunk-ui in commit freifunk-berlin/firmware-packages@dea32fadee2b69cc554d1690972f1860e1b5058b

As far as I can tell, it works.  But if someone has a better way to do this, I am totally willing to learn.  The documentation of the luci dispatcher is somewhat limited and very out of date.

SvenRoederer, 2019-12-17:
@pmelange can you give a rough commit at when you reference "luci-master"? 
Was it before branching opwnwrt-18.06?

It seems to me that this issue is also mentioned in https://github.com/freifunk-berlin/firmware/issues/735

pmelange, 2019-12-17:
https://github.com/openwrt/luci/issues/2273 should be a good start