freifunk-berlin/firmware #616: feeds.conf: comments about additional feeds would be useful

pmelange, 2018-11-11:
I recommend adding comments to feeds.conf.  Here is my proposal:

```
###
# OpenWrt
#
src-git packages https://github.com/openwrt/packages.git^327a03e8f0c61f8ee98adbee5bd37ce7bccbb049
src-git luci https://github.com/openwrt/luci.git^5ec72617eede467c932642b60291d627fca70342
src-git routing https://github.com/openwrt-routing/packages.git^1b9d1c419f0ecefda51922a7845ab2183d6acd76
###
# Freifunk Berlin
#   packages_berlin: addons, defaults, uplinks, utils
#
src-git packages_berlin https://github.com/freifunk-berlin/firmware-packages.git^d21ec055dc7ce199837f60e323639e9b4ca4054a
###
# Addtional
#   gluon: l3roamd, simple-tc
src-git gluon https://github.com/freifunk-gluon/packages.git^c34d129afb48f225741b683127beee6ba3c169d2

##
# just for reference (syntax of different feed-sources)
# see https://wiki.openwrt.org/doc/devel/feeds for more info
##
#src-git packages_berlin https://github.com/freifunk-berlin/firmware-packages.git;master
#src-link packages_berlin ../../../firmware-packages
```

SvenRoederer, 2018-11-11:
probably it's more easy and up-to-date, when just looking in the packages-directory of a build to see which feeds provides what package.

pmelange, 2018-11-11:
none of the files in the packages directory have l3roamd or simple-tc.

pmelange, 2018-11-11:
nothing gluon here either:  http://buildbot.berlin.freifunk.net/buildbot/stable/1.0.1/ar71xx-generic/packages/packages/mips_24kc/

pmelange, 2018-11-12:
I still think that it makes sense to have comments in the feeds.conf file.  Having this information readily available without having to compile is documentation.  Needing to compile to find out what is happening is not documentation.  

Sure, the information could be placed somewhere else, like in a buried commit message, but that is still not the obvious place for (new) developers to look.