freifunk-berlin/firmware #617: olsrd LinkQualityAlgorithm in Profile does not work for community profile of Potsdam

niccokunzmann, 2018-11-13:
I build and configured many routers using Berlin Firmware Release Hedy [v1.0.1](https://github.com/freifunk-berlin/firmware/tree/v1.0.1) the last days.

<details>
<summary>
Its <a href="https://github.com/openwrt/luci/blob/master/contrib/package/community-profiles/files/etc/config/profile_potsdam#L34">profile</a> version uses etx_ffeth.
</summary>
Hedy v1.0.1 pulls the feed from luci.
I can see the file <a href="https://github.com/openwrt/luci/blob/master/contrib/package/community-profiles/files/etc/config/profile_potsdam#L34">firmware/openwrt/feeds/luci/contrib/package/community-profiles/files/etc/config/profile_potsdam</a>
with the source code

```
config 'defaults' 'olsrd'
	option 'LinkQualityAlgorithm' 'etx_ffeth'
```

</details>

However, for each router, I needed to configure the link quality algorithm myself because it was set to etx_ff.

I would like to investigate further into this, [inform the community](https://lists.freifunk-potsdam.de/pipermail/users/2018-November/018624.html) and update this issue.

References:
- Moved from https://github.com/openwrt/luci/issues/2276

SvenRoederer, 2018-11-18:
the relevant code is here .... https://github.com/freifunk-berlin/firmware-packages/blob/v1.0.1/utils/luci-app-ffwizard-berlin/luasrc/tools/freifunk/assistent/olsr.lua#L24

pmelange, 2018-12-21:
Once PR https://github.com/freifunk-berlin/firmware-packages/pull/174 has been merged, this problem should be fixed.

niccokunzmann, 2019-09-17:
Now, when I use the upstream-1907/master branch, I see that this is fixed. Should I close this now or do you need it still?

closed 2019-09-17 by niccokunzmann

SvenRoederer, 2019-09-17:
the master-branch of the firmware is still on 18.06, so it's not fixed there. The upstream-1907 + master branches are not official and somehow just my "fun" (see https://github.com/freifunk-berlin/firmware/issues/583)