freifunk-berlin/firmware #619: 5GHz interface does not join mesh on

robertfoss, 2018-11-14:
Router: TP-Link Archer C59 v1
Build: `LuCI openwrt-18.06 branch (git-18.212.55608-5ec7261) / Freifunk Berlin 1.1.0-alpha 16a3e40 `

### Expected behaviour
5Ghz radio starting being brought up and joining the local 5Ghz mesh.

### Actual behaviour
5Ghz radio is recognized, but the interface is not brought up or joins the mesh.

[image]
[image]

```
# cat /sys/kernel/debug/ieee80211/phy1/ath9k/dfs_stats
DFS support for macVersion = 0x600, macRev = 0x0: disabled
Pulse detector statistics:
    pulse events reported    :          0
    invalid pulse events     :          0
    DFS pulses detected      :          0
    Datalen discards         :          0
    RSSI discards            :          0
    BW info discards         :          0
    Primary channel pulses   :          0
    Secondary channel pulses :          0
    Dual channel pulses      :          0
Radar detector statistics (current DFS region: 2)
    Pulse events processed   :          0
    Radars detected          :          0
Global Pool statistics:
    Pool references          :          0
    Pulses allocated         :          0
    Pulses alloc error       :          0
    Pulses in use            :          0
    Seqs. allocated          :          0
    Seqs. alloc error        :          0
    Seqs. in use             :          0
```

```
# iwinfo 
wlan1-adhoc-2 ESSID: "intern-ch13.freifunk.net"
          Access Point: 70:4F:57:26:63:DC
          Mode: Ad-Hoc  Channel: 13 (2.472 GHz)
          Tx-Power: 18 dBm  Link Quality: 33/70
          Signal: -77 dBm  Noise: -93 dBm
          Bit Rate: 1.0 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1

wlan1-dhcp-2 ESSID: "berlin.freifunk.net"
          Access Point: 72:4F:57:26:63:DC
          Mode: Master  Channel: 13 (2.472 GHz)
          Tx-Power: 18 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -93 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1
```

```
# iw list
Wiphy phy1
	max # scan SSIDs: 4
	max scan IEs length: 2257 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0x7 RX 0x7
	Configured Antennas: TX 0x7 RX 0x7
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
		 * outside context of a BSS
	Band 1:
		Capabilities: 0x11ee
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 8 usec (0x06)
		HT TX/RX MCS rate indexes supported: 0-23
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm)
			* 2472 MHz [13] (20.0 dBm)
			* 2484 MHz [14] (disabled)
	valid interface combinations:
		 * #{ managed } <= 2048, #{ AP, mesh point } <= 8, #{ P2P-client, P2P-GO } <= 1, #{ IBSS } <= 1,
		   total <= 2048, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz }

	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
Wiphy phy0
	max # scan SSIDs: 16
	max scan IEs length: 199 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
	Band 2:
		Capabilities: 0x19ef
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 8 usec (0x06)
		HT TX/RX MCS rate indexes supported: 0-15
		VHT Capabilities (0x339979b2):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			TX STBC
			SU Beamformer
			SU Beamformee
			MU Beamformer
			MU Beamformee
			RX antenna pattern consistency
			TX antenna pattern consistency
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 0 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 0 Mbps
		Frequencies:
			* 5180 MHz [36] (20.0 dBm)
			* 5200 MHz [40] (20.0 dBm)
			* 5220 MHz [44] (20.0 dBm)
			* 5240 MHz [48] (20.0 dBm)
			* 5260 MHz [52] (20.0 dBm) (radar detection)
			* 5280 MHz [56] (20.0 dBm) (radar detection)
			* 5300 MHz [60] (20.0 dBm) (radar detection)
			* 5320 MHz [64] (20.0 dBm) (radar detection)
			* 5500 MHz [100] (26.0 dBm) (radar detection)
			* 5520 MHz [104] (26.0 dBm) (radar detection)
			* 5540 MHz [108] (26.0 dBm) (radar detection)
			* 5560 MHz [112] (26.0 dBm) (radar detection)
			* 5580 MHz [116] (26.0 dBm) (radar detection)
			* 5600 MHz [120] (26.0 dBm) (radar detection)
			* 5620 MHz [124] (26.0 dBm) (radar detection)
			* 5640 MHz [128] (26.0 dBm) (radar detection)
			* 5660 MHz [132] (26.0 dBm) (radar detection)
			* 5680 MHz [136] (26.0 dBm) (radar detection)
			* 5700 MHz [140] (26.0 dBm) (radar detection)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (13.0 dBm)
			* 5765 MHz [153] (13.0 dBm)
			* 5785 MHz [157] (13.0 dBm)
			* 5805 MHz [161] (13.0 dBm)
			* 5825 MHz [165] (13.0 dBm)
			* 5845 MHz [169] (13.0 dBm)
	valid interface combinations:
		 * #{ managed } <= 1, #{ AP, mesh point } <= 16,
		   total <= 16, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz }

	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	Device supports VHT-IBSS.
```

pmelange, 2018-11-15:
The problem is that the driver for the newer 5Ghz devices do __not__ support adhoc.  There are a couple workaround which you could do.

1) Change the 5Ghz adhoc interface to 802.11s, set the mesh id to "Mesh-Freifunk-Berlin" (repeat on all other devices that mesh at 5Ghz).
2) Delete the adhoc wireless interface.  This means that meshing at 5Ghz will no longer work.  Additionally you would have to remove the wirelessX interface from Network/Interfaces and OLSR4 and 6.  With this setup, the client AP at 5Ghz will start working.

There should be a discussion about which driver to use.  There are currently two firmware drivers.  Both have their advantages and disadvantages.  There is the original driver (which doesn't support adhoc) and the -ct driver [1] which supports adhoc but not encrypted AP's when there are vif's (needed for a user who would like to have their private lan extended with their freifunk router [2]).

This issue is related to https://github.com/freifunk-berlin/firmware/commit/491da5533425e87b10d2e91c2e397775d3f243e3#commitcomment-31278661

[1] http://www.candelatech.com/ath10k-10.1.php
[2] https://wiki.freifunk.net/Berlin:erweiterte_Konfiguration#Weiterleiten_des_privaten_Wlan

SvenRoederer, 2018-11-15:
alternatively check with the recent SAm0815_experimental build, which sets up 802.11s by default and don't deal with AdHoc anymore.

robertfoss, 2018-11-15:
I just tried both, and they do seem to bring up .11s interfaces, but with
two routers configured the same way, I'm not sure they're seeing each other.

>

pmelange, 2018-11-16:
Just checking, but you are using a separate set of ip addresses for each node, right?

Also, on the command line, you can run 'neigh.sh' to see if the node is meshing. Or take a look at the olsr status pages on the web interface.

robertfoss, 2018-11-16:
```
# neigh.sh
Local Remote vTime LQ NLQ Cost Host 

Local               Remote             vTime  LQ       NLQ      Cost           Host 
fd2e:28e1:693d:2::1 2001:bf7:831:63::1 121718 0.310000 0.000000 4194304.000000 
```

pmelange, 2018-11-16:
The quality of the connection looks poor and I see only ipv6 meshing. How far are they from each other? 

Did you use a different set of ip addresses for each router? Would you please post your /etc/config/network files for both routers?

robertfoss, 2018-11-18:
```
host3:~# neigh.sh
Local       Remote       vTime  LQ       NLQ      Cost     Host 
10.31.42.74 10.31.36.247 138177 0.886000 0.788000 1.430664 host2.olsr 

Local               Remote              vTime  LQ       NLQ      Cost           Host 
fd0e:f68c:df44:2::1 fd2e:28e1:693d:2::1 139240 0.886000 0.732000 1.538086        
fd0e:f68c:df44:2::1 2001:bf7:831:63::1  121404 0.345000 0.000000 4194304.000000
```

```
host3:~# cat /etc/config/network 

config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd0e:f68c:df44::/48'

config interface 'wan'
	option proto 'dhcp'
	option peerdns '0'
	option type 'bridge'
	option ifname 'eth0 ffuplink_wan'

config interface 'wan6'
	option ifname 'eth0'
	option proto 'dhcpv6'
	option peerdns '0'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '1 2 3 4 0t'

config interface 'tunl0'
	option ifname 'tunl0'
	option proto 'none'

config rule 'olsr_allif_ipv4'
	option lookup 'olsr'
	option priority '1000'

config rule 'localnets_allif_ipv4'
	option lookup 'localnets'
	option priority '2000'

config rule 'olsr_tunnel_tunl0_ipv4'
	option lookup 'olsr-tunnel'
	option priority '19999'
	option in 'tunl0'

config rule 'olsr_default_tunl0_ipv4'
	option lookup 'olsr-default'
	option priority '20000'
	option in 'tunl0'

config rule 'olsr_default_unreachable_tunl0_ipv4'
	option action 'unreachable'
	option priority '20001'
	option in 'tunl0'

[... 187 lines cut ...]

config rule 'olsr_tunnel_dhcp_ipv4'
	option lookup 'olsr-tunnel'
	option priority '19999'
	option in 'dhcp'

config rule 'olsr_default_dhcp_ipv4'
	option lookup 'olsr-default'
	option priority '20000'
	option in 'dhcp'

config rule 'olsr_default_unreachable_dhcp_ipv4'
	option action 'unreachable'
	option priority '20001'
	option in 'dhcp'

config device 'ffuplink_dev'
	option name 'ffuplink'
	option macaddr '4a:fd:53:58:e0:39'

config interface 'ffuplink'
	option ifname 'ffuplink'
	option proto 'dhcp'
	option ip4table 'ffuplink'
```

pmelange, 2018-11-18:
Did you set the 5Ghz device on both nodes to 802.11s?  I have a similar setup working here (but with different hardware).  5Ghz is meshing with 802.11s and 2.4Ghz is meshing with adhoc.

Could you post /etc/config/wireless from both devices?

robertfoss, 2018-11-18:
```
host2:~# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option channel '36'
	option path 'pci0000:00/0000:00:00.0'
	option disabled '0'
	option country 'DE'
	option doth '0'
	option hwmode '11acn'
	option diversity '1'
	option htmode 'HT20'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/qca956x_wmac'
	option htmode 'HT20'
	option disabled '0'
	option country 'DE'
	option doth '0'
	option hwmode '11bgn'
	option diversity '1'
	option channel '13'

config wifi-iface
	option network 'wireless0'
	option device 'radio0'
	option bssid '02:36:CA:FF:EE:EE'
	option ssid 'intern-ch36.freifunk.net'
	option mcast_rate '12000'
	option mode 'adhoc'
	option ifname 'wlan0-adhoc-5'

config wifi-iface
	option ifname 'wlan0-dhcp-5'
	option network 'dhcp'
	option encryption 'none'
	option device 'radio0'
	option mode 'ap'
	option ssid 'berlin.freifunk.net'

config wifi-iface
	option network 'wireless1'
	option encryption 'none'
	option device 'radio1'
	option bssid 'D2:CA:FF:EE:BA:BE'
	option ssid 'intern-ch13.freifunk.net'
	option mcast_rate '6000'
	option mode 'adhoc'
	option ifname 'wlan1-adhoc-2'

config wifi-iface
	option ifname 'wlan1-dhcp-2'
	option network 'dhcp'
	option encryption 'none'
	option device 'radio1'
	option mode 'ap'
	option ssid 'berlin.freifunk.net'
```

```
host3:~# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option channel '36'
	option path 'pci0000:00/0000:00:00.0'
	option disabled '0'
	option country 'DE'
	option doth '0'
	option hwmode '11acn'
	option diversity '1'
	option htmode 'HT20'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/qca956x_wmac'
	option htmode 'HT20'
	option disabled '0'
	option country 'DE'
	option doth '0'
	option hwmode '11bgn'
	option diversity '1'
	option channel '13'

config wifi-iface
	option network 'wireless0'
	option device 'radio0'
	option bssid '02:36:CA:FF:EE:EE'
	option ssid 'intern-ch36.freifunk.net'
	option mcast_rate '12000'
	option mode 'adhoc'
	option ifname 'wlan0-adhoc-5'

config wifi-iface
	option ifname 'wlan0-dhcp-5'
	option network 'dhcp'
	option encryption 'none'
	option device 'radio0'
	option mode 'ap'
	option ssid 'berlin.freifunk.net'

config wifi-iface
	option network 'wireless1'
	option encryption 'none'
	option device 'radio1'
	option bssid 'D2:CA:FF:EE:BA:BE'
	option ssid 'intern-ch13.freifunk.net'
	option mcast_rate '6000'
	option mode 'adhoc'
	option ifname 'wlan1-adhoc-2'

config wifi-iface
	option ifname 'wlan1-dhcp-2'
	option network 'dhcp'
	option encryption 'none'
	option device 'radio1'
	option mode 'ap'
	option ssid 'berlin.freifunk.net'
```

pmelange, 2018-11-18:
5Ghz on both devices are set up to use adhoc.  adhoc does not work with the current driver.  Please change to 802.11s with MeshID "Berlin-Freiufnk-Mesh" and see if it work better.

robertfoss, 2018-11-18:
That seems to work! So, this required manual configuration and I'm still not able to route packets onto the internet.

This is from a 3rd node, called host1.
```
host1:~# neigh.sh
Local       Remote       vTime  LQ       NLQ      Cost     Host 
10.31.42.77 10.31.42.74  136515 0.831000 1.000000 1.202148 mid1.ManteuffelStr91-3.olsr 
10.31.42.76 10.31.42.73  141380 1.000000 1.000000 1.000000 ManteuffelStr91-3.olsr 
10.31.42.77 10.31.36.247 136793 1.000000 0.894000 1.118164 mid1.ManteuffelStr91-2.olsr 
10.31.42.76 10.31.36.246 142671 0.886000 1.000000 1.127930 ManteuffelStr91-2.olsr 

Local               Remote              vTime  LQ       NLQ      Cost     Host 
fd45:a34a:929a:2::1 fd0e:f68c:df44:2::1 138336 0.894000 0.894000 1.250000  
fd45:a34a:929a:1::1 fd0e:f68c:df44:1::1 137108 1.000000 1.000000 1.000000  
fd45:a34a:929a:2::1 fd2e:28e1:693d:2::1 130683 0.831000 1.000000 1.202148  
fd45:a34a:929a:1::1 fd2e:28e1:693d:1::1 138386 1.000000 1.000000 1.000000
```

robertfoss, 2018-11-18:
> Yup, typo...
> 
> I don't know what you mean by "can't join the public mesh".
The ad-hoc berlin freifunk mesh, as far as I understand it 802.11s does not mesh with the ad-hoc net.

> 
> This is a developer's test image. Not everything works automagically yet. So, sometimes a bit of manual configuration is necessary. :)
Of course! Just wanted to be overly clear!

> 
> I don't think your third point is related to #580. But it is also not related to this issue. Could you open up a new issue? Maybe also show a traceroute from a connected client (not on the uplink node). Also, could you show the results of "ip tun" on all of your nodes?
Ack, created #623