freifunk-berlin/firmware #620: Mesh to internet forwarding not working

robertfoss, 2018-11-14:
Router: TP-Link Archer C59 v1
Build: `LuCI openwrt-18.06 branch (git-18.212.55608-5ec7261) / Freifunk Berlin 1.1.0-alpha 16a3e40 `
Tunnel: Yes
Network topology: 
```
client <-> freifunk node <-> dhcp <-> interwebz
                         <-> tunnel <-> interwebz
```

# Expected results
Connecting a laptop to the berlin.freifunk.net SSID and obtaining and IP should allow me to connect to
the actual internet.

# Actual results

The tunnel interface seems to come up on the FF node, but no traffic is router through it.
When using a non-tunnel version of this image, I'm seeing the same issue, where no traffic is routed out to the actual internet.

```
client $ ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
```

```
freifunk# ip route get 8.8.8.8
8.8.8.8 via 192.168.0.1 dev br-wan src 192.168.0.4 
    cache
```

```
freifunk# ip link show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-wan state UP mode DEFAULT group default qlen 1000
    link/ether 70:4f:57:26:63:dd brd ff:ff:ff:ff:ff:ff
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP mode DEFAULT group default qlen 1000
    link/ether 70:4f:57:26:63:dc brd ff:ff:ff:ff:ff:ff
4: tunl0@NONE: <NOARP,UP,LOWER_UP> mtu 1480 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1
    link/ipip 0.0.0.0 brd 0.0.0.0
5: ifb0: <BROADCAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc hfsc state UNKNOWN mode DEFAULT group default qlen 32
    link/ether 52:46:b8:d8:60:f0 brd ff:ff:ff:ff:ff:ff
6: ifb1: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 32
    link/ether fe:d6:93:dc:4a:9d brd ff:ff:ff:ff:ff:ff
9: br-dhcp: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether 70:4f:57:26:63:dc brd ff:ff:ff:ff:ff:ff
10: eth1.1@eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-dhcp state UP mode DEFAULT group default qlen 1000
    link/ether 70:4f:57:26:63:dc brd ff:ff:ff:ff:ff:ff
11: br-wan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether 70:4f:57:26:63:dd brd ff:ff:ff:ff:ff:ff
12: wlan1-adhoc-2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether 70:4f:57:26:63:dc brd ff:ff:ff:ff:ff:ff
13: wlan1-dhcp-2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-dhcp state UP mode DEFAULT group default qlen 1000
    link/ether 72:4f:57:26:63:dc brd ff:ff:ff:ff:ff:ff
607: ffuplink: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc hfsc state UNKNOWN mode DEFAULT group default qlen 100
    link/none
```

```
freifunk# ip route show
default via 192.168.0.1 dev br-wan proto static src 192.168.0.4 
10.36.226.240/28 dev br-dhcp proto kernel scope link src 10.36.226.241 
172.31.240.0/24 dev ffuplink proto kernel scope link src 172.31.240.5 
192.168.0.0/24 dev br-wan proto kernel scope link src 192.168.0.4
```

robertfoss, 2018-11-15:
The same behaviour has been replicated on the firmware/SAm0815_experimental branch: 0a16b523d46a6d9febef5c17daf91c675ab1953f
`LuCI Master (git-18.296.76718-9deb5c1) / Freifunk Berlin Dev-alpha 0a16b52 `

pmelange, 2018-11-15:
Thanks for reporting this issue, but it is a duplicate of #580.  I'm closing this issue so that the focus stays on the other issue.

pmelange, 2018-11-15:
P.s. Maybe give the tunneldigger image a try.  (the alpha-test-tunneldigger branch).  An announcement for alpha-testers will hopefully be sent out on the mailing list in the next couple days.

closed 2018-11-15 by pmelange

robertfoss, 2018-11-15:
> P.s. Maybe give the tunneldigger image a try. (the alpha-test-tunneldigger branch). An announcement for alpha-testers will hopefully be sent out on the mailing list in the next couple days.

Will do! Thanks for the hint.