freifunk-berlin/firmware #621: Luck feed: the web interface to opkg is now it's own package

pmelange, 2018-11-16:
luci-mod-system no longer does opkg management. It has been split out into a new package called luci-app-opkg.  The upstream commit is openwrt/luci@ aa2e0e2

Should we add luci-app-opkg to common.config or make it a dependency of luci-mod-freifunk-ui?

booo, 2018-11-26:
Does luci-mod-freifunk-ui need the package to work or is it more of an optional plugin? If it's optional please add it to the common.config

pmelange, 2018-11-26:
luci-mod-freifunk-ui does not __need__ the new luci-app-opkg.  I'll get a PR out with the changes soon (common.config and all the files in the packages directory).

SvenRoederer, 2019-11-15:
fixed with 0f9db4f751505ab7591c78bd3de35cf6623b0dcd

closed 2019-11-15 by SvenRoederer