freifunk-berlin/firmware #623: Only able to route to internet from uplink node, not mesh nodes

robertfoss, 2018-11-18:
Router: TP-Link Archer C59 v1
Build: `LuCI luci-app-olsr-new-json-lib branch (git-18.315.67843-5fd51f0) / Freifunk Berlin Dev 1.1.0-alpha 151ac8c `
Configurations:
```
laptop <> mesh <> node1 <> uplink
laptop <> node2 <> mesh <> node1 <> uplink
```
# Expected behaviour
Both connection through meshed and directly uplinked node should route to the internet.

# Actual behaviour
The connection through node1 routes to the internet, but the connection through node2 does not.

# Debug

```
1:~# ip tun
tunl0: any/ip remote any local any ttl inherit nopmtudisc
```

```
2:~# ip tun
tunl0: any/ip remote any local any ttl inherit nopmtudisc
tnl_0a1f2a4c: ip/ip remote 10.31.42.76 local any ttl 64
```

```
laptop$ traceroute 8.8.8.8
traceroute to 8.8.8.8 (8.8.8.8), 30 hops max, 60 byte packets
 1  frei.funk (10.36.178.209)  0.445 ms  0.472 ms  0.552 ms
 2  * * *
 3  * * *
 4  * * *
 5  * * *
 6  * * *
 7  * * *
 8  * * *
 9  * * *
10  * * *
11  * * *
12  * * *
13  172.31.228.190 (172.31.228.190)  2848.179 ms !H  2848.234 ms !H  2848.204 ms !H
```

SvenRoederer, 2018-11-19:
is this probably related to #550 ?

pmelange, 2018-11-19:
It seems like OLSR is having a problem writing to the routing tables:
```
Mon Nov 19 11:10:36 2018 daemon.info olsrd[2126]: Received netlink error code File exists (-17)
Mon Nov 19 11:10:36 2018 daemon.err olsrd[2126]: Error on inserting policy rule aimed to activate RtTable 112!
```

When I do ```ip r g 8.8.8.8 from 10.36.147.114 iif tunl0```, on the uplink router, I get
```
RTNETLINK answers: Network unreachable
```

The rules on a 1.0.1 router are totally different than on a dev router.

1.0.1 (uplink type no-tunnel)
```
~# ip rule
0:	from all lookup local 
1000:	from all lookup olsr 
2000:	from all lookup localnets 
19989:	from all to 192.168.200.0/24 iif tunl0 prohibit
19989:	from all to 192.168.200.0/24 iif wlan0-adhoc-2 prohibit
19989:	from all to 192.168.200.0/24 iif wlan1-mesh-5 prohibit
19989:	from all to 192.168.200.0/24 iif br-dhcp prohibit
19989:	from all to 192.168.200.0/24 iif eth0.100 prohibit
19990:	from all iif tunl0 lookup ffuplink 
19990:	from all iif wlan0-adhoc-2 lookup ffuplink 
19990:	from all iif wlan1-mesh-5 lookup ffuplink 
19990:	from all iif br-dhcp lookup ffuplink 
19990:	from all iif eth0.100 lookup ffuplink 
19999:	from all iif tunl0 lookup olsr-tunnel 
19999:	from all iif br-dhcp lookup olsr-tunnel 
19999:	from all iif eth0.100 lookup olsr-tunnel 
19999:	from all iif wlan1-mesh-5 lookup olsr-tunnel 
19999:	from all iif wlan0-adhoc-2 lookup olsr-tunnel 
20000:	from all iif tunl0 lookup olsr-default 
20000:	from all iif br-dhcp lookup olsr-default 
20000:	from all iif eth0.100 lookup olsr-default 
20000:	from all iif eth0.100 lookup olsr-default 
20000:	from all iif wlan1-mesh-5 lookup olsr-default 
20000:	from all iif wlan0-adhoc-2 lookup olsr-default 
20001:	from all iif tunl0 unreachable
20001:	from all iif br-dhcp unreachable
20001:	from all iif eth0.100 unreachable
20001:	from all iif wlan1-mesh-5 unreachable
20001:	from all iif wlan0-adhoc-2 unreachable
32766:	from all lookup main 
32767:	from all lookup default 
100000:	from all lookup olsr-tunnel 
100010:	from all lookup olsr-default 
```

dev (uplink type no-tunnel)
```
~# ip rule
0:	from all lookup local 
1000:	from all lookup olsr 
2000:	from all lookup localnets 
10000:	from 192.168.100.30 lookup ffuplink 
19999:	from all iif tunl0 lookup olsr-tunnel 
19999:	from all iif br-dhcp lookup olsr-tunnel 
19999:	from all iif eth0.100 lookup olsr-tunnel 
19999:	from all iif wlan0-mesh-2 lookup olsr-tunnel 
20000:	from all iif tunl0 lookup olsr-default 
20000:	from all iif br-dhcp lookup olsr-default 
20000:	from all iif eth0.100 lookup olsr-default 
20000:	from all iif wlan0-mesh-2 lookup olsr-default 
20000:	from all to 192.168.100.30/24 lookup ffuplink 
20001:	from all iif tunl0 unreachable
20001:	from all iif br-dhcp unreachable
20001:	from all iif eth0.100 unreachable
20001:	from all iif wlan0-mesh-2 unreachable
32766:	from all lookup main 
32767:	from all lookup default 
90012:	from all iif lo lookup ffuplink 
100000:	from all lookup olsr-tunnel 
100010:	from all lookup olsr-default 
```

pmelange, 2018-11-19:
@robertfoss Can you please post the contents of /etc/config/ffwizard?

I noticed that in my file, the sharenet variable was set to 0.  It should be set to 1.

robertfoss, 2018-11-19:
```
# cat /etc/config/ffwizard

config settings 'settings'
	option sharenet '1'
	option usersBandwidthUp '20'
	option usersBandwidthDown '250'
	option enableStats '0'
	option ssid 'berlin.freifunk.net'
	option dhcpmesh '10.36.226.224/28'
	option meship_radio0 '10.31.42.73'
	option meship_radio1 '10.31.42.74'
	option runbefore 'true
```

pmelange, 2018-11-19:
I have just set up two nodes, with 802.11s mesh, using the no-tunnel "default" image.  Everything works fine.  Could you imagine resetting both devices and running the wizard again?

(by the way, there is also a new alpha-test-tunneldigger image).

pmelange, 2018-11-19:
I just repeated the above with the tunneldigger image (for the uplink router) and everything works fine too.

pmelange, 2018-11-19:
for the nodes, are you using no-tunel (default) or tunneldigger?  Maybe there is some strange edge case that needs to be worked on.  I'm busy playing with as many possible configurations as possible to try to find out what does and does not work.

robertfoss, 2018-11-19:
I've testing both, but am using tunneldigger and it seems to work. I don't know why it wasn't working at first.

pmelange, 2018-11-19:
Ok. I ran into a situation where the packages were not being forwarded.

This is the situation.  I set up two uplink nodes (it doesn't matter if it's no-tunnel or tunnel).  On the node that I was the client on, I pulled the cord from the wan port.  The node needed 13 minutes to recognize that the wan connection was dead.  During these 13 minutes, the router that I was connected to was not making DNS requests correctly (because it thought that the default route was going through the wan port).

With no-tunnel, also all client traffic failed to be forwarded to the other node (even with giving an IP addr such as 8.8.8.8).  With tunneldigger, the tunneldigger service noticed after about 1 minute that the connection had been lost, and would send packets to the remaining uplink node (as long as I didn't need to do any DNS queries).

After the 13 minutes, everything worked just fine.  Good thing that my lease time is short.  With a longer lease time, it could have taken weeks.

I wonder if this is similar to your situation.

@SvenRoederer Do you think we should have some kind of ping watchdog for the wan port?  The hotplug script doesn't react when the wan port is on a switch.  I also tried pulling the wan cable on a device where the wan port is not on the switch, and the hotplug script ran (almost) immediately.

pmelange, 2018-11-19:
These two threads might help point us in the right direction:

https://forum.openwrt.org/t/hotplug-d-script-is-not-called-after-eth0-unplug/7533

https://forum.openwrt.org/t/hotplug-doesnt-handle-wan-link-down/6685/6

PolynomialDivision, 2018-12-09:
Do you disabled smart gateway?

robertfoss, 2018-12-09:
No.

pmelange, 2019-06-27:
This has been fixed in https://github.com/freifunk-berlin/firmware-packages/pull/182

Please try the 1.1.0-rc2 release candidate and see if you can reproduce this issue.  If so, please reopen

closed 2019-06-27 by pmelange