freifunk-berlin/firmware #625: ffwizard: individual texts for uplinks

SvenRoederer, 2018-11-19:
the wizard has some hardcoded texts which points to a specific uplink (eg VPN03 - https://github.com/freifunk-berlin/firmware-packages/pull/155/commits/eb9d6c4c8fe35777936e3d1fb5a1d8b904a1eb50#diff-0019160469d0108641e7020d912d2336L45). 
We should add a way to customize these text-blocks by the installed uplink.

I see 2 possible ways:
* the uplink package provides some files which directly are read by the wizard 
* the uplink-package sets some uci-options during it's installation, which are evaluated by the wizard

pmelange, 2018-11-24:
Check out https://github.com/freifunk-berlin/firmware-packages/commit/ef9fe6e95a5f73e350f2f5e65be325818b22e48d

pmelange, 2018-11-24:
Sorry, that is the diff you provided above (different commit id though).  All references are anyhow removed.  I still think it should be enough to have the needed info (and links) in the package description.

SvenRoederer, 2018-11-25:
got it ... missed that the uplink.auth=x509 also disables the messagebox I had seen.

closed 2018-11-25 by SvenRoederer