freifunk-berlin/firmware #630: mirgation: openvpn should be disabled with all upgrades

pmelange, 2018-11-26:
Irregardless of the version number being upgraded to, the init script for openvpn should be disabled.

Currently only an upgrade to 0.2.0 disabled openvpn.  Anything after leaves the server enabled.

SvenRoederer, 2018-12-02:
What's the problem in disabling or not disabling openvpn?

SvenRoederer, 2018-12-02:
disabling OpenVpn on all upgrades will cause loss of bbb-vpn connection and there will be no chance to enable it again from remote.

pmelange, 2018-12-02:
The hotplug script takes care of that.  The wizard disables it too.  One of the upgrades disabled it.  It should generally be disabled and the hotplug script should start it.

The problem is that if the router starts and meshes before it is plugged into the wan port, then openvpn will connect though the smart gateway.

SvenRoederer, 2021-06-13:
## general issue
To reenable all services is a upstream default of OpenWrt. Working on this (for some other issue / project) I published a [patch on the Mailinglist](http://lists.openwrt.org/pipermail/openwrt-devel/2021-January/033068.html). As this change is not covering all cases it was not accepted yet, but should do for us. 

## this specific issue
As we don't provide any OpenVPN-ish images anymore by default, this specific issue will not happen anymore.