freifunk-berlin/firmware #631: Wizard: third field at password request

Akira25, 2018-11-26:
After flashing the router with a Image of the master-branch the wizard requests for the passwords of the router. Using the tab key, the cursor doesn't switch to the confirmation field but to a third one. This one is located outside of the monitor, left-hand side.

SvenRoederer, 2018-12-19:
to which wizard are you referencing to?

pmelange, 2018-12-20:
I found this issue on the master branch.  After filling out the first field on the changePassword page, and hitting tab, the cursor goes off the page to the left.  Hitting tab again brings you to the second password field. 

I wonder if this is related to https://github.com/openwrt/luci/issues/2332