freifunk-berlin/firmware #632: German Law (draft): Requirements for a secure Broadband Router

pmelange, 2018-11-26:
The German government released "Requirements for a secure Broadband Router" which they plan on turning into law.

I have only found this document in English:
https://www.bsi.bund.de/SharedDocs/Downloads/DE/BSI/Publikationen/TechnischeRichtlinien/TR03148/TR03148.pdf;jsessionid=01F54E80B004E9BFB194DBC00DE9B961.2_cid360?__blob=publicationFile&v=2

Here are some things that might impact Freifunk.
* 3.1.2 for a community-wifi: ```Connection to other devices connected to the LAN interface, private WiFi or guest WiFi and/or access to the configuration of the router as described in Section 4.1.1: User Access to Configuration MUST NOT be allowed by the router.``` The configuration of the router, in our case, is primarily on the community-wifi.
* 4.1.1 ```In factory setting the router MUST allow end-user access to the configuration only using the LAN or WiFi interface.``` If you know what you are doing, you can do this via the WAN port on our firmware.  And it continues: ```If the router allows to access the configuration over the WAN interface (e.g. Webserver, App) as a customization feature this communication MUST be encrypted using TLS (e.g. HTTPS) using cipher suites fulfilling the recommendations in [TR-02102-2] Section 3: Recommendations and this feature MUST be deactivated in factory setting.``` This is also not our default.
* Table 6: ```The router MUST allow the end-user to retrieve information about the last or more login attempt(s)``` That should be an easy feature.  Maybe even something for upstream.
* 4.7 ```The router MUST NOT forward inbound IPv6 traffic, IF it does not belong to a known connection``` WTF?  Why the hell not?  It doesn't say why not.  There's nothing in the document about forwarding IPv4 traffic.  I'd like to understand the reasoning.

What does everyone think about this?

EG-freifunk, 2018-11-27:
Thanks for pointing to this. Was anybody from any Freifunk-Community involved, made comments or suggestions?

Some thoughts:
- The first issue is, that this Guideline applies mainly to IAP(WAN)-LAN-Routers, whereas an FF router is usually connected only through a virtual network with the Internet. FF is not at all a guest or community wifi offered by the home router, but is provided by a separated device, the freifunk router. Yes, according to 3.2 the requirements also apply to a freifunk device, but nevertheless the freifunk network is clearly (physically) separated from the home LAN.
- Concerning 4.1.1: As I understood it up to now, the "factory settings" of an FF router do not allow access through the WAN interface but only via LAN to 192.168.42.1. There is even no WiFi in factory settings, or am I being wrong here?
- And note that it says for the configuration over HTTPS only SHOULD and not MUST (see page 16):
"IF the router offers configuration through a web interface (local website provided by the router and accessed with a browser by the user) login to access the configuration SHOULD be secured using HTTPS and only cipher suites recommended in ..."
The text above cited by @pmelange applies to routers to be customized by the end-user ("as a customization feature"). So we should define at least, what is initialization or customization of the freifunk router.
- Chapter 4.1.2: But who is the end-user in the freifunk network? Anybody or only the driving-by user? Why the Login Attempts or the connected devices must be shown to anybody? Strange requirements. So we need here an additional user group ("_unknown participants_" or _guests_) with separate requirements.
- What about the Firmware Updates (chap. 4.2): "The manufacturer of the router MUST provide information on how long firmware updates fixing common vulnerabilities and exposures ... will be made available." Certainly here is not the manufacturer adressed, but the freifunk firmware developers. But what they will say? Forever  ;-)

booo, 2018-11-27:
I don't see how this implicates firmware development right now. It's not part of German law at the moment.

If you think we should try to influence the law making process please initiate a discussion on the general mailing lists (wlan-talk, berlin@, ...).

pmelange, 2018-11-28:
@booo my focus was at look at what possible changes we could possibly have to make.  (if it becomes law, if it applies to custom firmware, if we decide to abide by the law).

@EG-freifunk There are two ways to access the router via the WAN port on an unconfigured or configured ff router.  Either you let it get a DHCP address and you use that or you attach via ipv6 link-local (based on the MAC).  The first is easy to disable, the second one I'm not so sure.  By default it's in plant-text http.  It would be possible to set up uhttpd to be encrypted by default.  Unfortunately the certs are not trusted which could cause confusion.

in 4.1.1, for LAN and Wifi the word SHOULD is used but for WAN the word MUST is used.

I believe 4.1.2 relates to users who have logged in as administrator on the router.  The info regarding attached clients is already in Luci (on the status page you will see the leases, o the routes page's arp table you will possibly see there any static addresses).  Adding a simple "Last login was DD.MM.YY HH:MM:SS" on the status page should be easy to implement.

pmelange, 2019-03-12:
A related article: https://tech.slashdot.org/story/19/03/11/1835239/eus-plan-to-ban-sale-of-user-moddable-rf-devices-draws-widespread-condemnation