freifunk-berlin/firmware #633: please adding Taarget ipq40xx from open wrt

RobinBrandt87, 2018-12-02:
Ich hab keine Ahnung wie ich ein Target aus dem openwrt in in die Freifunkfirmware übertrage. Könnte mir das jemand zeigen bzw. das iqp40xx Packet in einen Branch keine alpha-test-tunneldigger oder ähnlich adden ? Die Dinger scheinen gut zu sein hab schon eine Fritte 4040 mit open wrt am laufen 802.11s laüft auch auf 5Ghz mit VHT80. 

Dankeschön

SvenRoederer, 2018-12-02:
in https://lists.berlin.freifunk.net/pipermail/berlin/2018-November/038439.html habe als Beispiel für ein neues Target auf https://github.com/freifunk-berlin/firmware/commit/696871104f90b7e99da6d0293cd2f7035d5f20e4 verwiesen.

pmelange, 2018-12-03:
I have created a branch "target_ipq40xx-generic" with a first attempt to get this target working.  For some reason it's not working.  But it is a good starting point for @RobinBrandt87 to continue with.

If you want to build on an experimental branch, do the following:
git checkout <branchname>   # for example alpha-test-tunneldigger
git merge origin/target_ipq40xx-generic
make TARGET=ipq40xx-generic

Good luck.  And if you get the files working, please add commits to the target_ipq40xx-generic branch.

SvenRoederer, 2018-12-17:
if this fixed by PR #644 and PR #645 ?

SvenRoederer, 2019-04-18:
any progress?
What's the problem by the way?

RobinBrandt87, 2019-04-18:
There is no Problem with this Target it works good with the openwrt18.06.02 from the master branch. The ipq806x works to. It’s easy to add this Images and I am tested it on a Fritzbox4040 and Glinet B1300.
I am building and hosted the firmware  for myself, because sometimes I need packages from the openwrt for example the vsftpd ore something to use an external usb storage.

Yours faithfully Robin

SvenRoederer, 2019-04-18:
@RobinBrandt87 beside kernel-modules you can always install (with opkg) packages from the original OpenWrt binary-builds

SvenRoederer, 2019-04-18:
I just saw, that support for this target has been added to the master branch in https://github.com/freifunk-berlin/firmware/commit/3d5fa62f2cc585211d10cc1214e08611e13feb2d. As @RobinBrandt87 mentioned that there is no general problem with this target, so we can close this issue?

SvenRoederer, 2019-04-18:
I opened PR https://github.com/freifunk-berlin/buildbot/pull/53 to enable automatic build by buildbots

sarumpaet, 2019-05-27:
https://github.com/freifunk-berlin/buildbot/pull/53 has been merged. Closing this.

closed 2019-05-27 by sarumpaet