freifunk-berlin/firmware #636: Hedy-1.0.2 dev: uplink tunnel-berlin (openvpn) not initalized properly

pmelange, 2018-12-10:
This is tested with build b76e14d0801bf359bc6d3b5eb87b4fb87819e38e

With the development version of the change-uplink code added to Hedy-1.0.2, the tunnel-berlin images are not working properly.

After running the wizard, /etc/config/ffwizard does not have any settings

Using default or tunnel-berlin-tunneldigger, /etc/config/wizard is set up properly

SvenRoederer, 2018-12-11:
I assume #637 is caused as of the same problem.

Is also the VPN03-flavor affected?

SvenRoederer, 2018-12-11:
just tested (with packages of branch "for-Hedy-1.0.2_change_uplink")
> root@frei.funk:/tmp# sysupgrade -n tunnel-berlin/hedy-1.0.2-sam0815.uplink-d40ceba-gl-ar150-sysupgrade.bin

directly after rebooting  I get this:

> root@gib-mir-einen-namen:(unreachable)/root# cat /etc/config/openvpn
config openvpn 'ffuplink'
        option enabled '0'
        option client '1'
        option dev 'ffuplink'
        option status '/var/log/openvpn-status-ffuplink.log'
        option up '/lib/freifunk/ffvpn-up.sh'
        option route_nopull '1'
        option proto 'udp4'
        option dev_type 'tun'
        option persist_key '1'
        option keepalive '10 60'
        option remote_cert_tls 'server'
        option comp_lzo 'no'
        option script_security '2'
        option cipher 'none'
        option mssfix '1300'
        list remote 'tunnel-gw.berlin.freifunk.net 1194'
        option ca '/etc/openvpn/tunnel-berlin-ca.crt'
        option extra_certs '/etc/openvpn/tunnel-berlin-extra.crt'
        option cert '/etc/openvpn/ffuplink.crt'
        option key '/etc/openvpn/ffuplink.key'

Which looks correct to me.
is exactly this procedure failing with the build you referenced?

pmelange, 2018-12-11:
Yes exactly

SvenRoederer, 2018-12-12:
@pmelange I just had a short test with "Hedy 1.0.2-alpha rev b76e14d" (tunnel-berlin flavor) and did not see the problem you stated above. Please recheck ...

pmelange, 2018-12-12:
I just retested flashing vpn03, tunel-berlin, and tunel-berlin-tunneldigger with b76e14d and I no longer have the issue stated above.

Strange...

closed 2018-12-12 by pmelange