freifunk-berlin/firmware #638: Hedy-1.0.2-dev: uci setting ffberlin-uplink.preset.previous is set multiple times

pmelange, 2018-12-10:
This is tested with build b76e14d

When performing multiple change-uplink procedures, the UCI value ffberlin-uplink.preset.previous has multiple entries.

pmelange, 2018-12-10:
example:

```
/etc/config# cat ffberlin-uplink 

config settings 'preset'
	option previous 'no-tunnel'
	option previous 'tunnelberlin_openvpn'
	option current 'tunnelberlin_tunnneldigger'

config settings 'uplink'
	option auth 'none'
```

pmelange, 2018-12-10:
After the next change-uplink to default, the ffberlin-uplink file looks like this

```
config settings 'preset'
	option previous 'tunnelberlin_openvpn'
	option previous 'tunnelberlin_tunnneldigger'
	option current 'no-tunnel'

config settings 'uplink'
	option auth 'none'
```

pmelange, 2018-12-12:
PR https://github.com/freifunk-berlin/firmware-packages/pull/170 has been created to implement the changes.  Feel free to merge at any time.

SvenRoederer, 2018-12-12:
merged in https://github.com/freifunk-berlin/firmware-packages/commit/adf37cd41b2dd9c4cb8277502f787c5f6d5e5e26

closed 2018-12-12 by SvenRoederer