freifunk-berlin/firmware #639: Hedy-1.0.2-dev: mac addr for ffuplink changes when change-uplink is performed

pmelange, 2018-12-10:
This is tested with build b76e14d

When changing, for example, between tunnel-berlin-tunneldigger and default (or visa-versa) the macaddr for ffuplink gets regenerated.  It would be better to have this stay between changes.  It might be a good idea to keep this value in ffberlin-uplink.

SvenRoederer, 2018-12-12:
see https://github.com/freifunk-berlin/firmware-packages/pull/171#issuecomment-446712769

SvenRoederer, 2018-12-20:
As the removal of the retired DNS-server has been backported to lede-17.01 we need to add a migration  for Hedy-1.0.2

SvenRoederer, 2018-12-20:
new DNS-Servers have been added to lede-17.01 (https://github.com/openwrt/luci/commit/27ff6e930f6b08560639ceebdc9595da311a9e10) which also needs to be migrated

pmelange, 2018-12-20:
Maybe this should be run with every migration.  Sort of a standard thing to do, always set the DNS servers.  https://github.com/freifunk-berlin/firmware-packages/blob/ea6d78598a3edf4daca5c72208946364b7d1b136/utils/freifunk-berlin-migration/uci-defaults/freifunk-berlin-01-migration.sh#L443-L452

I believe this is not related to the title of this thread.

SvenRoederer, 2018-12-20:
@pmelange correct my comments should have gone to https://github.com/freifunk-berlin/firmware/issues/551

probably oyu want to add your comment there also, then we can delete them here

SvenRoederer, 2018-12-20:
let's discuss this for hedy-1.1

torte71, 2018-12-22:
Why not deriving the ffuplink MAC-address from the interface, that it runs on (e.g. only set the first byte to 'fe' and take the other bytes from eth0)? No need to save it and it would always be the same, even after reflashing.

pmelange, 2018-12-22:
@torte71 It was also my first idea.  But after discussions at the freifunk meetings, where we spoke about ways to reduce the amount of data which can be linked back to a device, I thought it would be better to generate something random.

For hedy-1.1 I would like to get this integrated into the change-uplink code.  I feel that there are a lot more fine touches we could make to improve the change-uplink system.  Especially reducing duplicated code.

SvenRoederer, 2018-12-29:
I wonder if it will break anything, if we define a "ffuplink_dev" by default and rely on this. So we have always the same mac-address and need to carry this value in ffberlin-uplink.