freifunk-berlin/firmware #640: Hedy-1.0.2-dev: changing uplink should set the old uplink type to enable '0'

pmelange, 2018-12-10:
When changing a uplink, the configuration for the old uplink type should be disabled (option enable '0').  This is important because it could theoretically be possible to do the following:

install tunnel-berlin (openvpn)
change to default
install openvpn to connect with the old bbb-vpn

Currently this causes problems because the openvpn instance for the bbb-vpn will also try to connect for the ffuplink.  But if we set enable to 0, then no problem.

Another example:
install tunneldigger
change to default
install freifunk-berlin-bbbdigger

The tunneldigger software then tries to run on ffuplink since it is still enabled, but that is not what the user expects.

SvenRoederer, 2018-12-12:
Who will enable this option later again?

As bbb-vpn is manual setup only anyway, I don't think it's important for Hedy-1.0.x. For master-branch we should think of testing it.

pmelange, 2018-12-12:
I don't think you understood the example.  The issue is not with bbbvpn.  It is with potentially having openvpn installed, and tunneldigger installed, and both of them trying to make an ffuplink connection at the same time.

The option enable should be set to 0 when the previous uplink type needs it to be disabled.

When a new uplink needs to set the option to 1, it does it in the uci-defaults file.

SvenRoederer, 2018-12-13:
ffuplink-connection: is that defining the ffuplink-interface or using the ffuplink-interface?

Indeed, I probably do not see the problem.

pmelange, 2018-12-13:
OK.  
Example 1
When the software for openvpn is installed, and runs the wizard, the option in openvpn.ffuplink is enabled.  Then, when the user does a change_uplink to no-tunnel, the option openvpn.ffuplink is still enabled.  Now, the user installs openvpn to connect with the old bbb-vpn.  Openvpn starts everything that is enabled.  But the user wants to use no-tunnel.
* note, the reason the wizard has to set enable to 1 is because openvpn's init script is enabled by default.  The wizard both disables the init script and sets openvpn.ffuplink to enabled.

----
Example 2
When the user installs tunneldigger, tunneldigger.ffuplink is enabled by default.  Then the user installs no-tunnel.  Then the user installs the bbbdigger addon, which automatically installs the tunneldigger package and it's dependencies.  The hotplug script for tunneldigger still starts the ffuplink section because it is enabled.
* note, tunneldigger does not need to be enabled in the wizard because the init script is disabled by default.

SvenRoederer, 2018-12-17:
I feel bad of changing this in the v1.0.x series. It might break more that it will help.
We should think of this,and the possible changes required in the wizard, for the master-branch.

SvenRoederer, 2018-12-20:
So let's think of this for Hedy-1.1?

pmelange, 2018-12-21:
I feel like this is somehow related to https://github.com/freifunk-berlin/firmware/issues/603

SvenRoederer, 2019-01-13:
We are facing this problem in master too, right?

I think, there is only a low percentage of users, that might be affected by this issue. We should add a note to the Releasenotes of Hedy-1.0.2 and think of a fix in the master-branch. When this is done, we can backport this from master and push another Release on Hedy-1.0.x

pmelange, 2019-01-13:
Sounds like a good idea to me.