freifunk-berlin/firmware #649: RB750Gr-3 requires reflashing of the boot loader

SvenRoederer, 2018-12-22:
Just remember some post on the OpenWrt-devel list:
* http://lists.infradead.org/pipermail/openwrt-devel/2018-October/014443.html
* http://lists.infradead.org/pipermail/openwrt-devel/2018-August/013484.html
* http://lists.infradead.org/pipermail/openwrt-devel/2018-July/013323.html

As of the successful testreport I prefer using the code, which not requires reflashing the bootloader and having the board booting via it's native Routerboot. This will be more convenient for the enduser and also gives the chance of reverting to stock-fw.

When this is tested and we can prove it'S working, we should support the original author, to get it upstreamed.

SvenRoederer, 2018-12-22:
this PR https://github.com/openwrt/openwrt/pull/1184 links the code to support native boot. not sure if there is any difference to the above mentioned on patchwork

pmelange, 2018-12-23:
This is very interesting.  I do agree that native routerboot support is reall cool.  But I already flashed the bootloader on my device to uboot.  I haven't found a way to flash the bootloader back.  So supporting this change will also cause my device to become unusable/unsupported.

SvenRoederer, 2018-12-24:
I tested the code on board and it boots via netboot, just by holding the reset-button on power-up. So no reflashing of the bootloader required. So let's try to push this forward to upstream.

There was also some discussion about general way to deal with this situation in https://github.com/openwrt/openwrt/pull/1184. The current support in OpenWrt is based on the reflashed bootloader, which is not compatible to each other.
For now I suggest to remove this board from Hedy-1.0, till a solution was found upstream. So we will not encourage users to spread around these incompatible change even more.

SvenRoederer, 2018-12-30:
there is a new PR https://github.com/openwrt/openwrt/pull/1686 for pushing this upstream

SvenRoederer, 2019-01-14:
I assume it will work as you did on changing it to the different uboot

SvenRoederer, 2020-02-16:
**current firmware:**
As we have switched to OpenWrt-19.07 on master, we just skipped the release with "modified bootloader". So we have no "end-user" impact regarding this. (5327c9986da3a63b3d781f829e598e189898e4ea)

**reverting from 18.06 modified bootloader board-support:**
I reverted my board by using a recent OpenWrt-build. On this build I removed the "read-only" flag from the partitions in the boards-dts and booted the initramfs-image. I created matching images from my backup-data and flashed with mtd.

closed 2020-02-16 by SvenRoederer