freifunk-berlin/firmware #653: SAm0815-experimental: ffwizard does not finish correctly

everloop2, 2019-01-03:
ffwizard does not finish -> box says cause of developer mode

-> after manually finishing ffwizard (save apply changes) and box restart -> login as administrator -> ffwizard does show again and internet is not working for clients cause of missing /etc/config/ffwizard and ffuplink entry in /etc/config/qos

workaround, get internet working for clients:

- reset box
- disconnect box WAN  (if plugged some boxes act like switches and are not reachable due leasing ISP IP to clients)
-  connet to box LAN -> http://frei.funk -> Administration -> login (without password)
-  setup ffwizard -> save and reboot -> click on not saved changes (top right) -> save and apply
-  renew host IP lease (due new box IP)	-> reboot box (System -> reboot)
-  connet to box LAN http://frei.funk (ignore ffwizard showing up again)
-  set up custom LAN/WAN options if needed	-> for me: enable VLAN and set one LAN port to WAN, WAN proto -> ignore + bridge VLAN eth0.2 and ffuplink_wan
-  ssh root@frei.funk
-  vi /etc/config/qos	-> press [i] -> at # INTERFACES add:

		config interface ffuplink
			option enabled 1
			#10MBit downlink/1.2MBit uplink
			option download 11000
			option upload 1200
			option classgroup "Default"

- press [ESC] -> press [:] -> press [w][q] -> press [ENTER]
- vi /etc/config/ffwizard -> press [i] -> insert:

			config settings 'settings'
				option sharenet '1'
				option usersBandwidthUp '1.2'
				option usersBandwidthDown '11.0'
				option enableStats '1'
				option ssid 'XXXX.freifunk.net'
				option dhcpmesh 'XXXX'
				option meship_radio0 'XXXX'
				option runbefore 'true'

- press [ESC] -> press [:] -> press [w][q] -> press [ENTER]
- reboot now
- connect WAN

- addtional: if box WAN connected to CABLE modem, change ffuplink_dev MAC address to one ISP knows and likes

- vi /etc/config/network

		config device 'ffuplink_dev'
			option macaddr 'XXXX'

- additional: enable monitoring -> vi /etc/config/luci_statistics -> option enable '1'

SvenRoederer, 2019-01-03:
Please tell the git-hash of the build you are using.

The devmode is added by https://github.com/freifunk-berlin/firmware-packages/commit/6887ac95293df0770308241331201e444f0c1031 and can be disabled in "uci-config system"

the only thing, which might relate to uplink is "ffwizard.settings.sharenet". but I recently setup a node from scratch with build based on https://github.com/freifunk-berlin/firmware-packages.git^d17458804a2f02842e69f618c1f40fd75a08fdc9 and did not see such problems. Have you installed from scratch (sysupgrade -n)? It might also related to your special uplink-configuration.

everloop2, 2019-01-07:
build: freifunk-berlin-dev-sam0815-2ea4f03-arcadyan_arv752dpw22-sysupgrade

did install on a clean box (reset to defaults)

uplink was set up after ffwizard, wich did not finish correctly -> no automatic reboot, showing box is set to developer mode (dont know if its related to ffizard not finishing)

ill try resetting box and set devmode disabled

everloop2, 2019-01-07:
tested sam0815-2ea4f03 ffwizard with devmode disabled  -> /etc/config/system -> config settings 'ffberlin' -> option devmode '0'

- ffwizard finishes and box reboots
- password is set
- dhcp not set, stays at default 192.168.42.1
- some other files also not changed to settings entered in ffwizard -> /etc/config/ e.g. network, qos, olsrd, olsrd6..

pmelange, 2019-01-07:
> * dhcp not set, stays at default 192.168.42.1
> * some other files also not changed to settings entered in ffwizard -> /etc/config/ e.g. network, qos, olsrd, olsrd6..

@everloop2 This problems mentioned above will probably be fixed by https://github.com/freifunk-berlin/firmware-packages/pull/174

SvenRoederer, 2019-02-10:
As mentioned in comments it's work as designed or caused by other problems, I close this one.

Feel free to reopen

closed 2019-02-10 by SvenRoederer

everloop2, 2019-05-04:
clean build: Freifunk Berlin Dev-SAm0815 52ace39 / LuCI Master (f138fc93)

found it -> /firmware/packages/..

```
# GUI-basics
[...]
luci-mod-freifunk
```
master and other branches use: "luci-mod-freifunk-ui" package containing latest patches

using `luci-mod-freifunk-ui` does work as expected.

remember in older builds of Dev-SAm0815 sometimes luci-mod-freifunk did change to luci-mod-freifunk-ui and vice versa

SvenRoederer, 2019-05-04:
> remember in older builds of Dev-SAm0815 sometimes luci-mod-freifunk did change to luci-mod-freifunk-ui and vice versa

That's why this branch is named "experimental"