freifunk-berlin/firmware #661: rerunning the wizard should not break the setup

pmelange, 2019-02-04:
It seems that if a user re-runs the wizard, the setup breaks.  This should not be the case.  Instead the wizard should not overwrite and/or modify any customized settings.

SvenRoederer, 2019-12-31:
It would be really nice to make the Wizard work like this, but I think the Wizard is:
* a tool for initial configuration
* a tool for giving the normal user a common setup in some basic steps

For more advanced / custom configuration we provide the full LuCI-features and ssh-access. Based on this, we should ensure that:
* the wizard generates a common working  setup (contact, Monitoring, uplink, Mesh)
* the wizard can be used for changing some very common settings after initial setup (e.g. position, contact, shared bandwidth)
* the wizard is not modifying settings except the mentioned ones
* for other reconfiguration the user should resetup the node from factory-defaults, use the LuCI-UI or the scripts from the "freifunk-berlin-configtools"-package
* Options that should not be changed when rerunning the wizard should be readonly
* pages which entirely contain options not to change, should not be displayed at all

SvenRoederer, 2021-06-19:
As mentioned above and in the referencing issue making it (again to) reliable to rerun will be much appreciated. Just some needs to do so.