freifunk-berlin/firmware #662: switching from ar71xx to ath79 causes configuration issues

SvenRoederer, 2019-02-06:
The new target ath79 is planned to be the successor of ar71xx in the next OpenWrt-release and currently we already support more boards on ath79-generic than for ar71xx-generic. Upstream supports switching the to ath79 by simply using sysupgrade, but it's not ensured that all hardware-related options will be converted. So we will run into problems with config/wireless (hardware-path changes) and config/system (buttons and leds) might change.
Upstream suggests to start confirmation from scratch ("sysupgrade -n") to ensure proper setup.

Do we also want to go this way? Having the user to resetup it's node. Or do we want to try, to convert the current setup ?

SvenRoederer, 2019-02-06:
I can confirm, that going from ar71xx to ath79 at least gives a working wired network, as configured before, whereas the wifi gets a new "radio1" setup. This new radio is unconfigured, of course.

pmelange, 2019-02-07:
Wouldn't it be possible to do the following:

1. backup the wireless config ```cp /etc/config/wireless /etc/config/wireless-orig```
2. run the script which sets up wireless ```wifi config``` https://openwrt.org/docs/guide-user/network/wifi/basic#regenerate_configuration
3. translate the configured wireless interfaces from wireless-orig to the newly configured wireless

SvenRoederer, 2019-03-09:
there is also a gluon-issue dealing with this: https://github.com/freifunk-gluon/gluon/issues/1570

CodeFetch, 2019-07-29:
There need to be migration scripts on a per-device basis. Things like swapped eth0 and eth1 ports can happen. Another thing is that the new switch configs follow unwritten rules now. That means e.g. all switch ports are set to use tagged VLANs. It's a bad idea to just try to sysupgrade to ath79. I think there was some effort to add a migration script to OpenWrt, but I can't find any information about it.

SvenRoederer, 2020-02-16:
As 19.07 is out with official support for the ath79 target I suggest the following to reduce migration issues:
* let's provide factory-images only for ath79 as they are avail
* let's not provide sysupgrade-files for ar71xx-boards anymore

This will have users with fresh boards directly starting on ath79, but allows existing nodes to stay on ar71xx when updating the firmware. So new nodes will not experience possible migration-issues and for old nodes we can work on a proper solution till switching to OpenWrt 2020.x

This will require a bit code on the Makefile (for OpenWrt19.07 based releases) to remove the unwanted factory-images.