freifunk-berlin/firmware #665: dev-daily: unusual long access time for web interface

Akira25, 2019-02-14:
build: dev-daily-012a952 (2019-02-13) and dev-daily-1743bf7 
device: tplink_c50-v4

Accessing frei.funk, LuCI needs very long to (initially) load. It can take from 35 second up to a whole minute. After the first page (frei.funk) was load, the other pages load at normal speed.

pmelange, 2019-02-14:
I have created a branch called fix/hasRootPass which addresses this problem. I'm traveling now and will create a pull request when I am back on the 22nd. But if anyone else wants to make a pr for the branch in the meantime, feel welcome.

SvenRoederer, 2019-02-15:
in your commit https://github.com/freifunk-berlin/firmware-packages/commit/052a0a83f10c2b4be1feec54a8fc8ffcbcbdcc7e you are using `luci.sys.user.getpasswd("root")` which is not working.
>  logger(tostring(luci.sys.user.getpasswd("root"))) 
returns "nil" even when a password was set.
I'd choosen to add an additional CGI-script, as this runs a root (who has access to the password-file), whereas the LuCI is running as an unpriviledged user.

pmelange, 2019-02-21:
The cgi script is timing out where it is being called by LuCI.  When I run the same wget command manually, it returns immediately.  I suppose another way to fix this bug would be to get the wget call to work properly.

When it times out, the result not "password_is_set:no" which is not what https://github.com/freifunk-berlin/firmware-packages/blob/4ad95b8536653c6b4c61fc358ae6c778e4d6c92d/utils/luci-app-ffwizard-berlin/luasrc/tools/freifunk/assistent/tools.lua#L74 is checking for.

-----
While coding https://github.com/freifunk-berlin/firmware-packages/commit/052a0a83f10c2b4be1feec54a8fc8ffcbcbdcc7e, I tested the call luci.sys.user.getpasswd("root") differently.  I didn't convert it to a string.  I don't have the test code anymore, but it should look something like this:
```
if not luci.sys.user.getpasswd("root") then
  logger("passwd not set")
else
  logger("passwd set")
end
```

I don't guarantee that I'm right about this.  Also, I don't think I have the time to work on this the next couple of days.  I'll write another comment when I know more.

pmelange, 2019-02-23:
I just tested the call to luci.sys.user.getpasswd("root") as described in https://github.com/freifunk-berlin/firmware/issues/665#issuecomment-466071313.  The results are just as @SvenRoederer described.  It doesn't matter if the password has been set or not, the result is always the message "passwd not set".

So, now we need to find out why the call to wget is timing out.  Anyone have any ideas?

pmelange, 2019-02-24:
I tried changing the wget to /bin/wget and localhost to 127.0.0.1 and there was no change
When I run ps, I see the wget command, which simply times out causing our problem.

pmelange, 2019-02-24:
After working on a few more issue, I have tried the fix/hasRootPass branch again.  At the first boot, I get sent to the changePassword page.  I put in a password, and it works fine.  Afterwards I don't get confronted with the changePassword page again.  Page loading is at the normal speed.

Edge case:
If I instead decide not to enter a password on the changePassword page, then after going through the wizard and rebooting, the normal freifunk page comes up.  Many, but not all, of the pages in the interface show the built-in "No password set!" alarm box from OpenWrt.  If anyone finds it important to have the alarm box in this case be shown on all pages, then please open a separate issue.

SvenRoederer, 2019-02-24:
probably using luci.http can be used in place of calling wget. I used wget as I did not want to check for it's api (https://htmlpreview.github.io/?https://raw.githubusercontent.com/openwrt/luci/master/documentation/api/modules/luci.http.html)

pmelange, 2019-02-24:
I tried luci.sys.httpget and had the same results.  Then I looked into the code, and it does a wget.

PR #181 works with the current 18.06.x master branches.

SvenRoederer, 2019-02-24:
if I remember correctly, at least one of the problems fixed by the cgi-solution was:
* not displaying the wizard-password page, when the root-account had a password set by other means already (e.g. via a preseed uci-default script)

SvenRoederer, 2019-02-24:
> `luci.sys.user.getpasswd("root")`

see also https://github.com/freifunk-berlin/firmware/issues/208

SvenRoederer, 2019-02-24:
> So, now we need to find out why the call to wget is timing out. Anyone have any ideas?

Any better idea than bisecting?

pmelange, 2019-02-24:
Edge case 2:
Before loading the web interface, I logged via ssh and set a password.  Then when loading the web interface I was confronted with a request for a password.  The correct password doesn't seem to work and I was forwarded again to the same password request page.  (see description of https://github.com/freifunk-berlin/firmware/commit/8dc3728a1752f96153bf254952c18357d0fbf42f)
Solution:
Click on Administration at the bottom, log in, run the wizard from the menu.

This solution (with the two edge cases) is still better than the non-functioning cgi script.

SvenRoederer, 2019-02-25:
So we had a solution that was working. This solution broke by some change. Your solution has some known problems ...
So I suggest bisecting down to the point when it broke. there we will see what change and how to deal with that cause.

pmelange, 2019-03-02:
The results of bisecting (my oh my that takes a long time) is that the bug was introduced in https://github.com/freifunk-berlin/firmware/commit/2823ec3c3ae2563ac96df3b538808626bfc636b7

Now, I suppose it's time to go though all the feeds and bisect those too.  That will unfortunately take even more time.

But my concern is how https://github.com/freifunk-berlin/firmware/commit/2823ec3c3ae2563ac96df3b538808626bfc636b7 made it to the master branch.  It was not in a PR, and I don't think it was ever tested.  Isn't testing a prerequisite to pushing on the master branch?

pmelange, 2019-03-02:
I have finally found the root of the problem.  

The commit on openwrt/master is https://github.com/openwrt/openwrt/commit/c6aa9ff38870a30dbe6da17e4edad6039fe10ddf
the commit on openwrt/18.06 is https://github.com/openwrt/openwrt/commit/e5a0b6cde051bd24280e9c3a360e65617bce2416 (cherry-picked from master).

The effect is that the uhttpd config setting max_requests has been reduced from 3 to 1 for performance reasons.  But this causes the hasRootPass cgi-script to be blocked and it eventually times out.

There are potentially two solutions. 
* we override the setting ourselves.
* we convince upstream that it is a bad idea to have max_requests set to 1 and have the commit reverted.

I will create a PR for solution 1 since solution 2 may take a while.

pmelange, 2019-03-02:
The upstream issue is https://bugs.openwrt.org/index.php?do=details&task_id=2159

pmelange, 2019-03-02:
PR https://github.com/freifunk-berlin/firmware-packages/pull/183 has been created to solve this issue

SvenRoederer, 2019-03-02:
What about changing the cgi-script into an rpc-function? This will not cause the problems we see here with cgi-scripts.

pmelange, 2019-03-02:
rpc needs authentication https://github.com/openwrt/luci/wiki/JsonRpcHowTo.  In fact luci.sys.user.* functions are rpc functions and we know that they don't work.

I have spent far too much time on this because of a commit (https://github.com/freifunk-berlin/firmware/commit/2823ec3c3ae2563ac96df3b538808626bfc636b7) which was never tested before being commit to the master branch.  

If you have a better solution, please make a PR, get it peer reviewed, tested, and merged.  Then feel free to close PR https://github.com/freifunk-berlin/firmware-packages/pull/183

everloop2, 2019-03-15:
seems to be fixed at SAm0815, frei.funk and LuCI-login responds after 1-3 seconds

running on:
Modell:	Speedport W 504V Typ A (1x333Mhz)
Architektur:	Danube rev 1.5
Firmware Version:	Freifunk Berlin Dev-SAm0815 11afd3d / LuCI feed_mk branch (git-19.063.78882-af73045)
Kernel Version:	4.14.105

pmelange, 2019-03-19:
The web interface may be loading faster, but the functionality of hasRootPass is not yet ready.  Check out the comment in https://github.com/freifunk-berlin/firmware-packages/commit/5facfff6b0b585e7b0f509a13a7c2c205dc0b57b

I'm looking forward to seeing this work.

SvenRoederer, 2019-04-18:
please check https://github.com/freifunk-berlin/firmware-packages/pull/186. Seems to work ...

SvenRoederer, 2019-04-23:
upstream reverted the commit causing the seen issue, so it should be gone. But I think using rpcd is much more elegant than using the cgi we did till now.

Akira25, 2019-05-14:
I just tried it out. At least from users view, everything seems okay now.

closed 2019-05-14 by Akira25

SvenRoederer, 2019-05-15:
So I'll clean the code and merge freifunk-berlin/firmware-packages#186 some day.

Akira25, 2019-05-16:
Freifunk Berlin Dev-daily e805da7 works fine too. There is no long accces time and it gets the password request as usual.

pmelange, 2019-05-16:
Freifunk Berlin Dev-daily e805da7 pulls in a branch of the firmware-package (commit freifunk-berlin/firmware-packages@d2a9e5e) which does not have the changes in https://github.com/freifunk-berlin/firmware-packages/pull/186.  So it seems like the RPC code is not yet tested

SvenRoederer, 2019-05-16:
I did a test myself with 3e0000f639dee5cc274a3691956274f284d6e746 as it was fine. This build is based in OpenWrt before they reverted the uhttp-setting.