freifunk-berlin/firmware #674: olsrd is unmaintained, so we should avoid using it

SvenRoederer, 2019-03-31:
Olsrd is looking for a maintainer October 2018. This gives changing the core-routing software a new priority.
https://github.com/freifunk-berlin/firmware/projects/2#card-19601018

SvenRoederer, 2020-06-07:
By February the olsrd-v1 has gone to "hard freeze" because there was no new maintainer found (https://lists.olsr.org/pipermail/olsr-users/2020-February/007149.html)

SvenRoederer, 2020-08-02:
Even new maintainers have been found (https://github.com/OLSR/olsrd/pull/79, https://github.com/openwrt-routing/packages/pull/575) there will be very unlikely active development of new features.

Akira25, 2020-08-04:
> Even new maintainers have been found ([OLSR/olsrd#79](https://github.com/OLSR/olsrd/pull/79), [openwrt-routing/packages#575](https://github.com/openwrt-routing/packages/pull/575)) there will be very unlikely active development of new features.

Is there any feature you would like to have implemented? By the modular nature of OLSRv1 it should be rather easy to do so. As far as I can see, there is currently no stable routing protocol which provides the same features as OLSRv1 now, right?

Akira25, 2020-08-04:
In addition, as effectively Freifunk Berlin is now the maintainer of OLSR, it's future should be assured for a long time. Nevertheless we might have some discussions on other routing protocols. As this issue addresses specifically the maintenance of OLSR, which is resolved, I'd like to have this closed.

SvenRoederer, 2020-08-10:
By the words of the Headline of the issue, you are right. 
And yes, even being the maintainer doesn't makes olsr a better protocol.

closed 2020-08-10 by SvenRoederer

Akira25, 2020-08-12:
@SvenRoederer I disgree strongly on the implication, that OLSR is a bad routing protocol. In terms of scaling for example it's great and far better than batman.
We had a talk on meshing and some routing protocols on HdS last week on wednesday. If you want to have a discussion on a new routing protocol, you should maybe show up on some wednesday meetup at HdS,
There are quite a lot people, who can add really fundamental thoughts on this, like mwarning42 for example.

We can dedicate a whole evening on routingprotocol-discussions and announce it at the mailinglist.