freifunk-berlin/firmware #675: bbbdigger: unwanted output during package install

SvenRoederer, 2019-04-01:
I just installed the bbbdigger package and had following on terminal:
> Installing freifunk-berlin-tunneldigger (2017-12-13-d7db3500-1) to root...
Downloading http://buildbot.berlin.freifunk.net/buildbot/unstable/x86-generic/1184/packages/packages/i386_pentium4/packages_berlin/freifunk-berlin-tunneldigger_2017-12-13-d7db3500-1_i386_pentium4.ipk
Configuring kmod-udptunnel4.
Configuring kmod-udptunnel6.
Configuring kmod-l2tp.
Configuring kmod-l2tp-ip.
Configuring kmod-l2tp-eth.
Configuring freifunk-berlin-tunneldigger.
/etc/init.d/tunneldigger does not have a START or STOP value
Configuring freifunk-berlin-bbbdigger.
**cfg164d76**
Starting tunneldigger "bbbdigger" on bbbdigger

I wonder on the "cfg164d76" thing

pmelange, 2019-04-01:
https://github.com/freifunk-berlin/firmware-packages/blob/5e5e75f82a9c0f389a55b4446bfc8c252a752bba/addons/freifunk-berlin-bbbdigger/files/postinst.sh#L65

changing the line to ```uci -q add ...``` doesn't change the output.  I think it would be a bad idea to redirect the output to null just in case ```uci add``` ever fails.

I do not think that this is a big problem.

SvenRoederer, 2019-04-04:
When you are concerned about failing of "uci add" there should be something that checks the $RC of the call and then react properly.

SvenRoederer, 2020-04-07:
@Akira25 didn't you volunteer to take care of this in the todays meeting?

In addition: feel free to checkout the issues having the "good first issue" tag (https://github.com/freifunk-berlin/firmware/labels/good%20first%20issue)

SvenRoederer, 2020-04-10:
> I think it would be a bad idea to redirect the output to null just in case `uci add` ever fails.

In PR https://github.com/freifunk-berlin/firmware-packages/pull/201#issuecomment-611941119 it seems you "forgot" about your concerns.  When braking it down this PR also just adds a redirection without testing for failing.