freifunk-berlin/firmware #676: olsrd jsoninfo not sending http headers

niccokunzmann, 2019-04-03:
When I access http://10.22.254.111:9090/version
then I would like to see the headers CORS headers - or at least any header.

Source code of [olsrd jsoninfo plugin at 0.9.0.3](http://olsr.org/git/?p=olsrd.git;a=blob;f=lib/jsoninfo/src/olsrd_jsoninfo.c;h=7bb9d8604fabb93647b439e654777d504ae1d44d;hb=c6fbdafd11ef1d31cbbeab138317c3fdd6673d1a#l1422).

However, there are NO headers present, not event Content-Length!
```
$ wget -SO- http://10.22.254.111:9090/version
--2019-04-03 04:59:21--  http://10.22.254.111:9090/version
Connecting to 10.22.254.111:9090... connected.
HTTP request sent, awaiting response... 
Length: unspecified
Saving to: ‘STDOUT’

-                             [<=>                                     ]       0  --.-KB/s             {
  "systemTime": 1554260362,
  "timeSinceStartup": 458060464,
  "version": {
    "version": "olsr.org - 0.9.0.3-git_9a1fd3e-hash_bf1cb86e7bc284f8fbd340a5681e692d",
    "date": "2017-04-10 10:01:33",
    "host": "ubuntu"
  }
}

-                             [ <=>                                    ]     227  --.-KB/s   in 0s     

2019-04-03 04:59:21 (10.5 MB/s) - written to stdout [227]
```

```
$ curl -D headers http://10.22.254.111:9090/version > /dev/null ; cat headers
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   225    0   225    0     0    225      0 --:--:-- --:--:-- --:--:--  1800
```
Even Firefox does not show any headers.

Why are there no headers? How to go about debugging this?

pmelange, 2019-04-03:
@niccokunzmann This is an upstream issue, which also has been fixed with the jsoninfo plugin 1.1.

> * Why are there no headers?

I don't know, but it is probably one of the reasons for creating a new version of the plugin

> 
> * Can you confirm this on your devices?

This affects all routers with plugin version 0.0

> 
> * How to go about debugging this?

Since this has been fixed upstream, I don't see the need to focus any effort on it.

The OLSR Status app for Luci seems to be able to read in the JSON data just fine.  The code which works with plugin 0.0 is here https://github.com/openwrt/luci/tree/lede-17.01/applications/luci-app-olsr/luasrc/view/status-olsr

The OLSR Status app has been updated to work with plugin 1.1 https://github.com/openwrt/luci/tree/master/applications/luci-app-olsr/luasrc/view/status-olsr

If you write your planned map code in lua, it should be relatively easy to borrow from the existing code to scrape out the information you need.

closed 2019-04-03 by pmelange

niccokunzmann, 2019-04-03:
Thanks for replying. That gave me new ideas!
Found the solution here:
https://github.com/OLSR/olsrd/blob/v0.9.0.3/lib/jsoninfo/src/olsrd_plugin.c#L137
`{ .name = "httpheaders", .set_plugin_parameter = &store_boolean, .data = &http_headers }, //`

To enable http headers for the JSON Plugin of OLSR,
1. log into your router using
2. edit /etc/config/olsrd `vi /etc/config/olsrd`
3. insert (`i`) a line the line `        option httpheaders 'yes'` in this file at the configuration of the JSONPlugin. My part of the file looks like this:
    ```
    config LoadPlugin
            option accept '0.0.0.0'
            option library 'olsrd_jsoninfo.so.0.0'
            option ignore '0'
            option httpheaders 'yes'
    ```
4. close with ESC, :wq Enter
5. `reboot`

Now, you should be able to enjoy HTTP headers!
http://frei.funk:9090/config
```
HTTP/1.1 200 OK
Date: Wed, 03 Apr 2019 12:32:57 GMT
Server: OLSRD JSONInfo plugin
Connection: closed
Content-type: application/json
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET, POST, OPTIONS
Access-Control-Allow-Headers: Accept, Origin, X-Requested-With
Access-Control-Max-Age: 1728000
Content-length: 3126
Cache-Control: no-cache
```