freifunk-berlin/firmware #678: dnsmasq won't run after update in case of numerous dnsmasq options

Aldrighetti, 2019-04-09:
After upgrade to hedy 1.0.2 dnsmasq stopped working because of numerous options in 
`list dhcp_option '3,10.36.11.1 6,10.36.11.1 10.36.11.2'´.
After upgrade to hedy 1.0.2 from kathleen 0.3.0 dnsmasq stopped working because it requires now one option per line in /etc/config/dhcp
`list dhcp_option '3,10.36.11.1
 list dhcp_option  6,10.36.11.1 10.36.11.2'´
Would be nice if that can detected.

SvenRoederer, 2019-04-11:
@Aldrighetti  the shown options have been added by you, right?
From the code I can only find :
* https://github.com/freifunk-berlin/firmware-packages/blob/f4d32074dd526693470c37777faef40baa3d3fd1/defaults/freifunk-berlin-dhcp-defaults/uci-defaults/freifunk-berlin-dhcp-defaults#L26 - defaults: which sets dhcp-option "3" on lan interface, which will be deleted later
* https://github.com/freifunk-berlin/firmware-packages/blob/36abae832685cadbf45cc8082d53be7459389c85/utils/luci-app-ffwizard-berlin/luasrc/model/cbi/freifunk/assistent/wireless.lua#L252 - wizard: which sets dhcp-option "119" on dhcp interface
* https://github.com/freifunk-berlin/firmware-packages/blob/b60db2573d8d09988c6ef7727cd5402e4ff90b7d/utils/freifunk-berlin-wizard-backend/files/usr/lib/ffwizard.d/60-dhcp.sh#L35 - wizard2: which sets dhcp-option "119" on dhcp interface

As the default setup only uses 1 dhcp-option (see 2nd point) this have probably never caused failures before.

pmelange, 2019-04-12:
Manually edited changes, which cause problems after upgrading, has also been cited in issue https://github.com/freifunk-berlin/firmware/issues/651.

If the configuration from CC or BB do not work anymore with LEDE (17.x) and higher, then I think it makes sense to check for these things in the migration scripts.  Now we know that ```list dhcp_option``` no longer works unless split up into multiple entries.

I recommend a generic function in the migration script which will take a list option and break it into multiple lines.  This could then be used where needed.

SvenRoederer, 2019-04-12:
When this was a manual change, and confirmed by @Aldrighetti, we should:
1. check why this was made (probably it's a general good idea to implement this)
2. We should think of how to deal with manual changes which
2.1. interfer with our defaults
2.2. do not interfer with our defaults, but are also not upstream defaults (like in #651)

But as mentioned above, the wizard should be fixed to use a list in place of a plain option.

Aldrighetti, 2019-04-30:
@SvenRoederer Yes can confirm it, that was an manual change.