freifunk-berlin/firmware #680: OpenWrt disabled support for TP-Link WDR4900

SvenRoederer, 2019-04-18:
https://github.com/openwrt/openwrt/commit/324e94f31bfdcf9c2245303434b087b306de0262 disables the WDR4900 as the kernel becomes to big for the bootloader. So it becomes quite interesting ...
Is there anyone who has such device to test if we can reduce the size of our kernel that it maybe fits?

SvenRoederer, 2019-06-06:
Have a look at https://github.com/openwrt/openwrt/pull/1773, which seems to contain patches to fix the support.
This patches can be added to the "SAm0815_experimental" or even "daily/upstream-master" builds but somebody with real hardware needs to test.

pmelange, 2019-06-07:
I'm building right now with CHKDSK88's branch in the config.mk.  I'll report more later.

pmelange, 2019-06-07:
Test successful.  I didn't test anything more than booting, but booting works!  I also saw how the kernel image would be loaded twice.  YEAH!

pmelange, 2019-06-07:
I have posted the results and the bootlog to https://github.com/openwrt/openwrt/pull/1773#issuecomment-499867649

SvenRoederer, 2019-06-07:
@pmelange do you have the patch ready to include into the builds (till integrated upstream)?

pmelange, 2019-06-07:
No, I just used the branch from CHKDSK88 in the config.mk file.  Currently I'm working on something more important (taxes :hankey:) and maybe can get to it after the weekend.

But maybe the changes will be integrated in daily/upstream-master by then.

I have also committed a new branch called ```upstream-master-wdr4900-test``` which is using the branch from CHKDSK88.

pmelange, 2019-06-10:
merged upstream https://github.com/openwrt/openwrt/commit/c4fdd4979b7803cd018b7206f1ec53f57cc2d854

closing issue

closed 2019-06-10 by pmelange