freifunk-berlin/firmware #681: Batman-adv MTU

Aldrighetti, 2019-04-30:
After to hedy 1.0.2 from 0.1.0 i got the following error in kernel Logs:
`batman_adv: bat0: The MTU of interface br-fflan is too small (1500) to handle the transport of batman-adv packets. Packets going over this interface will be fragmented on layer2 which could impact the performance. Setting the MTU to 1532 would solve the problem.`
Obviously the mtu is configured to 1532 in /etc/config/network for br-fflan but not recognized. perhaps this is related to netifd?? [Referenz](https://dev.archive.openwrt.org/ticket/18869.html)

SvenRoederer, 2019-04-30:
have you checked https://wiki.freifunk.net/Berlin:Tutorial:Mesh_via_LAN#Bonuslevel:_B.A.T.M.A.N._aktivieren_via_ssh --> point 5.)

> config interface 'wireless0"
           option mtu '1532'

Aldrighetti, 2019-04-30:
There's something odd; my wireless Adhoc Interface got the mtu of 1532 but not my bridge br-fflan. I#m not if there are config/declaration changes and because of that my config isn#t working anymore

carlmatthiasbrunogeorg, 2019-05-01:
this seems to be a common issue with bridges, please see https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1399064

closed 2019-05-27 by sarumpaet