freifunk-berlin/firmware #682: Update pre-set Digitalcourage DNS servers to new ip

carlmatthiasbrunogeorg, 2019-05-01:
While it is welcome that we suggest independent dns servers in our freifunk firmwares, the currently used pre-set ip of Digitalcourage 85.214.20.141 will expire in a couple of months. We should revise and update the default dns server ips (i.e. for Digitalcourage 46.182.19.48 respectively 2a02:2970:1002::18 as stated [here](https://digitalcourage.de/support/zensurfreier-dns-server)).

SvenRoederer, 2019-05-02:
Can you please open an issue against https://github.com/freifunk/openwrt-packages, as the DNS-settings will be taken from the community-profiles

pmelange, 2019-05-03:
The packages olsrd-defaults and migration in the firmware-packages repo are also affected.

@carlmatthiasbrunogeorg thank you for pointing this out.

pmelange, 2019-06-18:
PR https://github.com/freifunk-berlin/firmware-packages/pull/191 has been submitted to address this issue in firmware-packages.
PR https://github.com/freifunk/openwrt-packages/pull/16 has been submitted for the 19.x branch
PR https://github.com/openwrt/luci/pull/2776 has been submitted for the 18.x branch

pmelange, 2019-06-19:
https://github.com/freifunk/openwrt-packages/pull/16 has been merged
https://github.com/openwrt/luci/pull/2776 has been merged

pmelange, 2019-06-19:
https://github.com/freifunk-berlin/firmware-packages/pull/191 has been merged

closed 2019-06-19 by pmelange