freifunk-berlin/firmware #683: SAm0815-experimental: ffwizard does not finish correctly due package luci-mod-freifunk

everloop2, 2019-05-04:
see: https://github.com/freifunk-berlin/firmware/issues/653#issuecomment-489335354

SvenRoederer, 2019-05-07:
I dont see, what a problem you want to describe ...

everloop2, 2019-05-09:
**branch:** freifunk-berlin-dev-sam0815-52ace39
**platform:** device-arcadyan-arv8539pw22 (Speedport w504v typ a), Fritz7312.. -> platform independent

flash firmware -> run freifunk wizard -> finish freifunk wizard (due developer mode no automatic save & reboot) -> click on "unsaved changes"

**issue 1** : at "unsaved changes" -> "ffwizard" entry (`/etc/config/ffwizard`) is missing

click on "save and apply" -> after manual reboot -> file `/etc/config/ffwizard` is missing

**issue 2** : if monitoring is set at freifunk wizard -> its not enabled in `/etc/config/luci_statistics`

**workaround:** at `/firmware/packages/.. .txt` -> `# GUI-basics `-> change `luci-mod-freifunk` to `luci-mod-freifunk-ui` -> build firmware again -> issues 1 & 2 are gone

Is there a reason why Sam0815 branch uses `luci-mod-freifunk` while other branches use `luci-mod-freifunk-ui`?

SvenRoederer, 2019-05-15:
I switched back to luci-mod-freifunk to participate from the upstream changes.

but the missing generation of uci ffwizard is interesting.

SvenRoederer, 2019-06-01:
@everloop2 can you check if the correct uci-files will be created, after disabling the developermode (```uci set system.ffberlin.devmode=0```) and running the wizard?

everloop2, 2019-06-15:
freifunk-berlin-dev-sam0815-cb0ed01-arcadyan_arv8539pw22-sysupgrade.bin

"devmode = 1" -> no major bugs so far -> working  :D

-> /etc/config/ffwizard		-> created + looks good
-> /etc/config/luci_statistics	-> monitoring is enabled if checked at ffwizard
-> mesh					-> working	-> without pingcheck package?! -> last build had to install pingcheck package to get mesh working
-> freifunk wizard has manually to be finished -> go to "unsaved changes" -> "save&apply"	-> manual reboot

"devmode = 0" -> bugged	 :/

-> freifunk wizard has manually to be finished -> go to "unsaved changes" -> got hang here, after reconnect there was nothing to "save&apply" (before around 84 changes)
-> /etc/config/ffwizard -> gets created -> only a "config settings 'settings'" line there, others missing
-> /etc/config/luci_statistics -> monitoring is enabled if checked at freifunk wizard
-> mesh -> not working -> AP & mesh network not created -> stays a openWRT default
-> DHCP -> IP of box stays 192.168.42.1, DHCP network stays LAN (br-lan) openWRT default

everloop2, 2019-08-10:
Seems to be fixed :D

Freifunk Berlin (Dev-SAm0815 rev dd2f75e) -> at 16.07.2019 did a build

closed 2019-08-10 by everloop2